LibreCAD/LibreCAD #1764: Librecad doesn't cancel 'Close'

NabilAKROUD, 2024-04-24:
When in 
File > Close > Cancel

It close the open file instead of canceling.

perepujal, 2024-04-27:
Reproduce as Start LibreCad with an empty drawing, draw a line, not save the drawing, hit File -> Close
A dialog "Closing Drawing" appears with three options, Save, Close, Cancel

One could assume that "Save" will save the drawing and close the file, "Close" would close the file without saving. and "Cancel" will not save the drawing nor close the file and will return to the drawing.

Currently, "Cancel"  closes the file as if you hit "Close"
HTH
Pere

perepujal, 2024-04-27:
Strange, If I click on the window manager close button I get the same dialog, but then "Cancel" works as expected.

dxli, 2024-04-28:
the action queue needs a review:

we currently keep suspended actions in a stack, but validation of those actions appears to be weak.

NabilAKROUD, 2024-05-09:
I made a short video showing that the [Cancel] button doesn't did what it should do. 
https://github.com/LibreCAD/LibreCAD/assets/32953485/eb01dd00-6fa6-4d90-ac61-4ff1a363defa

closed 2025-04-02 by NabilAKROUD