LibreCAD/LibreCAD #1767: Error: Import Error: error opening DXF/DWG file

kashcode, 2024-04-28:
### Expected behavior

Open file and not receive error

### Observed behavior

While opening file error is sown:
[image]

### Steps to reproduce or sample file

Simple open LibreCAD and from menu file -> open, open some file and error is shown.
[Product1.zip](https://github.com/LibreCAD/LibreCAD/files/15144124/Product1.zip)

### Operating System and LibreCAD version info
Windows 11 
LibreCAD-Installer-2.2.0.2.exe

dxli, 2024-04-29:
@kashcode 

I have no trouble opening the dxf files.

Could you try our latest builds at:

https://github.com/LibreCAD/LibreCAD/releases

kashcode, 2024-04-29:
Already tried multiple times. Might be there is some more detailed logs where can try find what causes this?

lordofbikes, 2024-04-29:
All files open fine with `2.2.0.2`.
The error is about `open`, means the system call, to open the file, fails. It is not about the content. There is no more info or log file.
That means, it is an issue with file permissions or location or naming.
Try to move the DXF files to another location.
Give more details in steps to reproduce, e.g. absolute path of files, type of drive, pen drive or network or local.

kashcode, 2024-04-29:
@lordofbikes thanks for the hint, I will check things related to permissions

kashcode, 2024-05-01:
Path is following `C:\Users\ABgaldniecība\OneDrive\Dators\DXF` I guess issue is in this char `ī`. 
When I create folder under `C:\dxf` and copy file into this folder file opens without any error

dongxuli-concord-design, 2024-05-01:
Did you type in the path? 

By the file dialog should be language ignorant

kashcode, 2024-05-01:
Error can be reproduced. 
I just now create folder `galdniecība`  and add dxf files
[image]

When try open file get error:
[image]

OS Name	Microsoft Windows 10 Pro
Version	10.0.19045 Build 19045

dongxuli-concord-design, 2024-05-01:
How did you open the file? More specific:

1, from command line in the same folder by typing in the file name only? 
2, from File > open, type in the complete path and file?
3, file > open> click on the file name;
4, from file explorer, drag the file to a running LibreCAD?

kashcode, 2024-05-02:
in my particular case this is:
3. file > open> click on the file name;

dxli, 2024-05-12:
I can reproduce this one, and it appears to be a dxfrw bug.

If confirmed, this should also be fixed in 2.2.1

dxli, 2024-05-12:
Fixed by https://github.com/LibreCAD/LibreCAD/commit/72902cbb2f0e9c2efc2d714fccd82bafbffc0579

Please help testing.

The issue is NOT related to the encoding of the file name.