LibreCAD/LibreCAD #1769: Please don't tag RC/pre releases with the same tag pattern as releases

alerque, 2024-05-05:
Arch Linux packager here.

The tag https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.1 has no way of identifying from the Git tag that it is not part of a release series. The GH Release notes has a title that suggests it is a release candidate not a release and the GH Release has a status label, but again to consumers of the Git interface these are not obvious. Additionally when you do go to cut a release now the actual tag you'd want for the release is already taken, and re-tagging introduces a whole host of problems for downstream packagers.

Instead please use some other tag pattern such as a suffix like [semver](https://semver.org/) uses so that the purpose and status of a tag is clear from the tag semantics itself.

Thanks.

dongxuli-concord-design, 2024-05-05:
2.2.1 should be a mistake.

The intended 2.2.1 branch is for a future 2.2.1 release stable release. 

The tag should be 2.2.1_rc1

I tagged the 2.2.1 branch as Draft and the master as latest.

dxli, 2024-05-05:
Looks like the 2.2.1 tag was from CI automatic tags. 

The tag "2.2.1" has been deleted, and the 2.2.1 branch is not set for automatic tag 2.2.1_rc1 now.

lordofbikes, 2024-05-07:
Things are a bit messed up currently :disappointed: 

Probably started with 7597e21 : `git describe` with `--tags` uses any tag.
The intention is to use only tags with a comment, thus `--tags` should not be used there.
So tags, which should be used for asset naming, like `2.2.1_rc1` must be created with a tag comment!
`2.2.2_alpha` has a comment, thus it is correct.
Automatic and internal used tags like `latest` have no comment and are thus not used for asset naming.

891d882 : the `latest` tag is just for `CI` and with `automatic_release_tag` updated to let `CI` keep up with the last successful build.
So I also recommend to restore this.

The current naming with `latest` in the assets name is irritating, because GitHub labels stable release versions as `Latest`.
The naming of `Continuous Build` should clarify, that the assets are `alpha`/`beta` versions and probably not safe for production use. 
[image]

A `Draft` release is not seen public, it can only be seen by organization members!
Thus `Release Candidate Builds` must be public too and will then also be shown as `Pre-release` to public.
That they are pre-release must be obvious from title and assets naming.
[image]

e4ccbe3 : Use `rc-build` or `pre-release` or `draft` for automatic tags in `CI`
As mentioned above, we use commented tags for assets naming. The `automatic_release_tag` is just used by `CI` to track latest successful build.
So a fix tag for `CI` does not need to update `build-all.yml` for new releases, just create a new commented tag and see assets use the new release tag for naming.

dxli, 2024-05-13:
Reset naming to:

2.2.1_rc

2.2.2_alpha

Auto tagging is limited to "latest"

closed 2024-05-16 by dxli