LibreCAD/LibreCAD #1770: Move and Copy commands should be two different commands

ezequielleonzybert, 2024-05-05:
Just an opinion. Choosing between moving or copying every time I want to move something is a waste of time to me. Would be much better if move command just move, with no pop-up window, and a new command for copy that just copy.
Thanks for reading.

lordofbikes, 2024-05-05:
You can use cut/copy/paste instead of Modify-Move to avoid the pop-up.
Just `CTRL-X`, `CTRL-C`, `CTRL-V` and the extra click to define the reference/insert point.
There are tool buttons in the Edit tool bar too.
See https://docs.librecad.org/en/latest/ref/menu.html?edit-1#edit-1

ezequielleonzybert, 2024-05-05:
> You can use cut/copy/paste instead of Modify-Move to avoid the pop-up. Just `CTRL-X`, `CTRL-C`, `CTRL-V` and the extra click to define the reference/insert point. There are tool buttons in the Edit tool bar too. See https://docs.librecad.org/en/latest/ref/menu.html?edit-1#edit-1

I get what you say. I'm doing like that now but the move command functionality is better than the edit comands ctrl+x ctrl+c and ctrl+v. I'm showing you why in this example:

https://github.com/LibreCAD/LibreCAD/assets/25140875/f90d5348-8ac8-4888-b6d5-50a5564586f5

try to do that with edit CTRL commands. You would need to first draw a horizontal line to get the intersection that the move command already gives you.

ezequielleonzybert, 2024-05-05:
And here is the same with a move and copy command as I suggest 
https://github.com/LibreCAD/LibreCAD/assets/25140875/5f244cbb-ca8f-48f0-8a79-d6df86d94e40

dxli, 2024-05-05:
First, we don't copy or simulate other softwares;
Second, I don't think there's any legal issue for us to implement copying/moving entities;
Third, we are not following concepts or API of other softwares; the famous Google vs Oracle case showed that implementation on open APIs can be justified as fare use;
Currently, we have no intention to implement based on any APIs, unless the APIs are compatible with open source licenses.

sand1024, 2024-05-08:
Well, recently I've added a new "duplicate" action - and it is already merged.  The major intent for that command - creation of entity's copies in a bit faster way and with better control of layers and attributes for duplicated entities. 

Probably it will be useful for your case.  

Here is a demo for that command https://youtu.be/MmQMJ6Uhmkc

dxli, 2024-05-18:
Workaround exists

closed 2024-05-18 by dxli