LibreCAD/LibreCAD #1771: program startup error

akcansoft, 2024-05-06:
Continuous Build: LibreCAD-latest-1-g1a470768-win64.exe
Windows 11

Program startup error. Program is not working
[image]

dxli, 2024-05-06:
I suppose the failure is not limited to win11.

Did anyone try the installer on older versions?

GSStnb, 2024-05-06:
Is this related?
   $ /opt/lc/LibreCAD-latest-1-g423ce8a4.AppImage
   /tmp/.mount_LibreCOffjPC/usr/bin/librecad: error while loading shared libraries: libQt6Svg.so.6: cannot open shared object file: No such file or directory

akcansoft, 2024-05-06:
now I downloaded and installed **LibreCAD-latest-1-g423ce8a4-win64.exe**. I wanted to run the LibreCAD program.
The same problem still persists.

dongxuli-concord-design, 2024-05-06:
For Linux appimage, the issue appears to be clear, need to deploy the qt SVG libraries;

For windows, need troubleshooting

jonadem, 2024-05-07:
Win11 also :
[image]

dxli, 2024-05-07:
Win64 installer is fixed with https://github.com/LibreCAD/LibreCAD/commit/494a9c96e8b756cd80a4f859020ebaf6f2c8f61e

Looks like Linux appimage is the trouble for qt-6. 6

jonadem, 2024-05-07:
I have the same issue with the new installer.
From this comment, it looks related to libstdc++
https://github.com/huihut/OpenCV-MinGW-Build/issues/18#issuecomment-1842858728

akcansoft, 2024-05-07:
same problem persists.
Windows 11
LibreCAD-latest-9-gc8a920a0-win64.exe
Program startup error. Program is not working
[image]

dongxuli-concord-design, 2024-05-07:
Sorry, looks like I made a mistake here. The issue was an incorrect qt version was specified in packaging.

dxli, 2024-05-07:
https://github.com/LibreCAD/LibreCAD/commit/c8a920a0501f40e86f3ab6edeed670425035c7ee

This commit was built with proper qt version

GSStnb, 2024-05-07:
c8a920a
   This commit was built with proper qt version

The latest relase (just downloaded) generates the same error:
$ /home/opt/LibreCAD-latest-9-gc8a920a0.AppImage
/tmp/.mount_LibreCHLbpmf/usr/bin/librecad: error while loading shared libraries: libQt6Svg.so.6: cannot open shared object file: No such file or directory

dxli, 2024-05-09:
Linux appimage is fixed by manually copying the qt6 libraries over. The building tool simply quit after detecting an empty qt profile 

```
2024/05/07 15:15:40 Detected Qt 6
2024/05/07 15:15:40 Offset of qt_prfxpath: 4809676
2024/05/07 15:15:40 Length of value of qt_prfxpath: 0
2024/05/07 15:15:40 qt_prfxpath: 
2024/05/07 15:15:40 Could not get qt_prfxpath
2024/05/07 15:15:40 Got empty qtPrfxpath, exiting
LibreCAD
```
The windows issue is still not clear to me, seekg() is part of going and the library is located and copied into the installer. 

```
SetOutPath: "$INSTDIR"
File: "D3Dcompiler_47.dll" [compress] 1381077/4173928 bytes
File: "LibreCAD.exe" [compress] 2937632/9295360 bytes
File: "Qt6Core.dll" [compress] 2274219/6717624 bytes
File: "Qt6Gui.dll" [compress] 3365104/10306224 bytes
File: "Qt6Network.dll" [compress] 533450/1643184 bytes
File: "Qt6PrintSupport.dll" [compress] 142782/403120 bytes
File: "Qt6Svg.dll" [compress] 122300/370864 bytes
File: "Qt6Widgets.dll" [compress] 2164331/6627504 bytes
File: "libgcc_s_seh-1.dll" [compress] 30982/75264 bytes
File: "libstdc++-6.dll" [compress] 484878/1957376 bytes
File: "libwinpthread-1.dll" [compress] 21338/53248 bytes
File: "opengl32sw.dll" [compress] 5475867/20639888 bytes
File: Descending to: "..\..\windows\generic\"
File: "qtuiotouchplugin.dll" [compress] 37911/96960 bytes
File: Descending to: "..\..\windows\iconengines\"
```

dxli, 2024-05-12:
fixed by https://github.com/LibreCAD/LibreCAD/commit/3f057099733586dbf5de2b08460a9702946edc90

The problem appears to be an issue due to the mingw-64 version with gcc-8.10 . Fixed by using mingw-64, gcc-11.20

closed 2024-05-12 by dxli

jonadem, 2024-05-14:
Confirmation that it works with [LibreCAD-2.2.1_rc2-1-g4cd3b138-win64.exe](https://github.com/LibreCAD/LibreCAD/releases/download/2.2.1_rc-latest/LibreCAD-2.2.1_rc2-1-g4cd3b138-win64.exe)
Thanks ! :-)