LibreCAD/LibreCAD #1775: --------------------------- LibreCAD.exe - Entry Point Not Found --------------------------- The procedure entry point _ZNSi5seekgESt4fposI9_MbstatetE could not be located in the dynamic link library C:\Apps\LibreCAD\LibreCAD.exe.  --------------------------- OK    ---------------------------

NathanSweet, 2024-05-08:
### Observed behavior
Installed, run LibreCAD.exe, got error:
```
---------------------------
LibreCAD.exe - Entry Point Not Found
---------------------------
The procedure entry point _ZNSi5seekgESt4fposI9_MbstatetE could not be located in the dynamic link library C:\Apps\LibreCAD\LibreCAD.exe. 
---------------------------
OK   
---------------------------
```

### Operating System and LibreCAD version info
Latest continuous build: c8a920a0501f40e86f3ab6edeed670425035c7ee
Windows 10, 22H2, 19045.2006

I tried the official stable 2.2.0 and it worked.

dongxuli-concord-design, 2024-05-09:
2.2.1 branch should work also. 

I am having trouble to get our package building for qt6 (latest)

NathanSweet, 2024-05-09:
I tried 2.2.1_alpha-164-g09328dc7 (from SourceForge, wasn't found on GH) and confirmed it works. Cheers!

dxli, 2024-05-12:
Fixed by https://github.com/LibreCAD/LibreCAD/commit/3f057099733586dbf5de2b08460a9702946edc90

closed 2024-05-12 by dxli