LibreCAD/LibreCAD #1780: double free or corruption (out)

hapees, 2024-05-11:
## Expected 
   When I open librecad with a filename-argument 
   I can hit any key.

## Actual  
When I open librecad w/o a filename-argument 
then hitting an unassigned keys causes 
**double free or corruption (out)**
**Aborted (core dumped)**

## To repeat 
always 
Start librecad w/o filename-argument, Shift-key, crash

## Env. 
Fedora 40,  6.8.9-300.fc40.x86_64 and lower versions.
Librecad:
Version: 2.2.1_alpha-483-g89f8dc4e
Compiler: GNU GCC 14.0.1
Compiled on: May 11 2024
Qt Version: 6.7.0
Boost Version: 1.83.0

_Thank You_

dongxuli-concord-design, 2024-05-11:
Does it happen with our GitHub appimages?

closed 2024-05-12 by dxli

hapees, 2024-05-12:
Dongxu, thank you for the propt response.