LibreCAD/LibreCAD #1782: Strange drawing painting in QT6-based version?

sand1024, 2024-05-12:
Hi! 

I've recently built the application with QT6, and noticed that visual representation of drawing was changed comparing to previous version. 

It's hard to describe, yet I'll try - based on generic impression, it seems that some additional alpha channel is used during drawing, or probably some blending other mode of working with graphics is used (which is default in QT6 and QT5).  

Visually, I've noticed that all drawing under QT6 version is softer. 

For the grid - with specific zoom levels some irregular structure of drawing becomes visible (I didn't checked grid drawing logic yet, but do not remember such a behavior with QT5).  Please refer to attached screen. 
[image]

Also, there is strange color for lines overlapping - previously, if there are two lines with different colors located in the same coordinates,  the color of top/last line was displayed. 

But now - it seems that on the screen there is some blended color... 
[image]
[image]

Again, not sure whether its' related to QT6 or my environment, please take a look. 

Version: 2.2.2-alpha
Compiler: GNU GCC 11.2.0
Compiled on: May 12 2024
Qt Version: 6.7.0
Boost Version: 1.84.0
System: Windows 10 Version 22H2

sand1024, 2024-05-12:
Hm... it's weird, yet I can't attach screenshots... can drop them by email if needed

dxli, 2024-05-12:
Probably some rendering optimization of qt6

closed 2024-10-19 by sand1024