LibreCAD/LibreCAD #1783: Saving dxf produces an unreadable file for AutoDesk Viewer

pretla-karl, 2024-05-13:
### Expected behavior

Files saved in LibreCAD are usable/viewable in Autodesk Viewer as that is a standard product and/or tool. 

### Observed behavior

Parsing of the file fails with an unrecoverable exit code, image of error in the zip file. I must admit that it somewhat does not concern LibreCAD, yet on the other hand the expectation would be that I would be able to export an usable file. 

### Steps to reproduce or sample file

[Issue archive.zip](https://github.com/LibreCAD/LibreCAD/files/15292323/Issue.archive.zip)

### Operating System and LibreCAD version info
macOS Sonoma 14.1.2 tested with both LibreCAD 2.2.0.2 and LibreCAD-2.2.1_rc2-1-g3c407e5c
Windows 10 Pro 19045.4291 tested with LibreCAD 2.2.0.2

Haibara-z, 2024-05-21:
Have you solved it?

lordofbikes, 2024-05-21:
This is hard to evaluate, if this is an _us_ or _them_ problem.

It turned out, that _Autodesk_ software throws _Error 50 in drawing header on line 930_ for the file in question. The line number may vary for other files.
Searching for this error shows a lot of similar results on _Autodesk_ forums.

The original file is AC1032 (ACAD 2018), LibreCAD saved in format AC1021 (ACAD 2007).
The only issue in line 930 is the value of the variable `$XCLIPFRAME`, which is `2`.
All DXF references, 2007 (R21) and 2018 (R32), even the latest 2023 (R37) list only `0` and `1` as valid values.
Thus the error in _Autodesk Viewer_ is raised by the value `2` of header variable `$XCLIPFRAME`.
Changing the value in line 930 to `1` opens the LibreCAD file in _Autodesk Viewer_.

LibreCAD just red the value `2` from the original file, and saved it with the same value in its latest format. There is no hint in DXF references about value `2` for variable `$XCLIPFRAME`.

So the questions are:
  * What is the source of `77771_ORIG.dxf`?
  * Why does _Autodesk Viewer_ accept this value for DXF format `AC1032` but not for `AC1021`?

I can't see any reliable solution from LibreCAD to solve this.
Especially, when the official DXF reference does not document this value `2`.
Because _Autodesk Viewer_ opens the original file in DXF 2018 format, it might be a hidden feature, added after DXF 2007. But then they should have a fallback solution to handle this value for older formats.