LibreCAD/LibreCAD #1787: Weird Screen->Drawing coordinates mapping with shift (QT6, master)

sand1024, 2024-05-17:
Hi!

I've just discovered strange behavior in QT6 based version (from master) for coordinates processing. 

It seems (at least, as I suspect) that there is some discrepancy in mapping screen coordinates from mouse event to inner graph coordinates... 

The side effect of such discrepancy - incorrect catch of entity (for example, in parallel action original line is selected only if mouse moves from one side from the line) and incorrect positioning of points as result of click (say, setting relative point, setting start point of line etc.). 

I've found such effect in free mode (no specific snaps).  It seems that there is some x and y shift between the point where mouse click is clicked and resulting entity... 

As far as I've noticed, and that's why I suspect that something is related to coordinates mapping - the value of the shift seems to be the same (in pixels, not in drawing units) for different zoom levels. 

Shift signs are +x, -y. 

What is even more strange for me - it seems that coordinates widget displays correct coordinates positions on mouse move. But when click occurs (say, for setting relative zero) - relative zero point is drawn in shifted position... 

I've did some quick debugging, yet not found the reason yet. May check this with more details later on, yet probably such or similar issue with positioning already occurred earlier? 

Anyway, please check this in your environment to ensure that this is not something specific and related to my local setup. 

### Expected behavior

Mouse click that creates some point in drawing (like start of line, re

### Observed behavior

Point is created with some shift from click point 
The following video illustrates that behavior - https://youtu.be/fiBFettr5Mc

### Steps to reproduce or sample file

Entity catch error
1) Create horizontal line 0,0-0,100
2) invoke Parallel command
3) try to select line moving mouse from top to bottom
4) try to select line moving mouse from bottom to top

Positioning error (not the click point and created point)
1) Try to set relative point in some position (say, 0,0)
2) try to set start point of line 

### Operating System and LibreCAD version info
Master branch
Version: LC_VERSION
Compiler: GNU GCC 13.1.0
Compiled on: May 17 2024
Qt Version: 6.7.0
Boost Version: 1.84.0
System: Windows 10 Version 22H2

dongxuli-concord-design, 2024-05-17:
The offset is independent of scaling, so the trouble is in mouse coordinate space

dxli, 2024-05-18:
The problem is QCursor.

We use our own custom cursor images, and the hotX/Y should be at cener or the cursor image.

In Qt6, the ctor of QCursor requires hotX/Y=-1 to use the center, instead of the lower-right cornor. 

Before e0c35b5, the cursor uses the lower right corner as the hotspot, leaving a mismatch from the image center. The commit changes the parameters hotX/Y to -1 to use put the hotspot on the cursor image center again.

GSStnb, 2024-06-12:
Re: Microscopic Mouse Cursor on HiRes Display #1765

I finaly have been able to test the appImage (Version: 2.2.2_alpha1-96-ge7526bb8) and the Microscopic Mouse Cursor issue has been resolved.  The mouse cursor is consistent no matter where it is within a LC window and appears to retains the system size.

sand1024, 2024-10-26:
fixed

closed 2024-10-26 by sand1024