LibreCAD/LibreCAD #1790: Solid hatch does not cover some elements even if it is pushed to top with Order

dellus, 2024-05-21:
### Expected behavior:
Solid hatch covers all elements below if it is pushed to top with Order.

### Observed behavior:
Polylines, text and dimension line text stay visible.

### Steps to reproduce or sample file:
Draw some elements including polylines, rectangles, dimension lines. Draw an extra rectangle covering large part of the other elements and fill it with a solid hatch of some colour. Push the solid to the uppermost position with the Order command.

### Operating System and LibreCAD version info:
Version: 2.2.1_rc2-1-g4cd3b138 Appimage
Compiler: GNU GCC 11.4.0
Compiled on: May 13 2024
Qt Version: 5.15.3
Boost Version: 1.74.0
System: fossapup64 9.5

dellus, 2024-05-21:
Sample file: 
[solidtest220.zip](https://github.com/LibreCAD/LibreCAD/files/15395734/solidtest220.zip)
An even more weird sample, an older file:
[arch_cm_100_a4_uebung_220.zip](https://github.com/LibreCAD/LibreCAD/files/15395746/arch_cm_100_a4_uebung_220.zip)
It shows right in normal view, but wrong in Print Preview. I couldn't reproduce this in a new drawing.
All samples behave well in LibreCAD 2.2.0. Not in 2.2.1.

dxli, 2024-05-21:
What about hatches

dxli, 2024-06-02:
For solid, we should allow setting of its alpha channel.

Or simply setting alpha=1, say, for printing.

dellus, 2024-06-02:
Alpha channel for solids - what does this mean? I know  in bitmap graphics it's about transparency. Would this help for this issue? It works all right in 2.2.0 without that feature.
But anyway I would appreciate to be able to set a transparency value for a solid. It's valuable for graphical effects, for example in an architectural facade drawing, laying a shade over an already coloured surface to give a three-dimensional impression.
This would make LibreCAD an interesting tool for graphic designers as it is a pain to draw with exact dimensions in usual graphic apps like inkscape. A new area for LibreCAD to expand into.

By the way, the new Entity Info tool does not react to a solid (normal hatch works), and the solid also prevents other entities beneath it to be detected.

dxli, 2024-06-06:
It appears to me the trouble is limited to polylines. 

Other primitive geometries are hidden by the solid fill, if sent to bottom

dellus, 2024-08-30:
I want to remind of this issue, as 2.2.1 final is nearing. Has there been any work on this?
As it is ok in 2.2.0 there should be a solution.

solidtest220.dxf in LC 2.2.0:
[image]

solidtest220.dxf in LC 2.2.1rc :
[image]

dxli, 2024-08-30:
I will see whether a fix is possible for 2.2.1.

Otherwise, since this is a problem limited to polylines, a workaround could be exploding polylines for printing, and wait for a fix in 2.2.2