LibreCAD/LibreCAD #1791: Package AppImage with Qt6 included

gerardbm, 2024-05-25:
Since Qt6 is a requirement, I think that the AppImages should have it packaged inside. Is that possible?

As I can see, older versions had a size about 46 MiB, while the last versions got reduced to 12 MiB. I can't test last LibreCAD versions.

dellus, 2024-05-25:
Same with me.

dxli, 2024-05-25:
Other than the size, does it work?

gerardbm, 2024-05-25:
It doesn't run in my case, since I don't have Qt6 in my system. I hope someone else can test it. Surely, it's only a packaging issue.

dxli, 2024-05-25:
The current appimage builder couldn't handle qt later than 5.14 well

I will try other builders

dxli, 2024-05-25:
fixed by manually copying all Qt6 libraries. This is probably an overkill, but it's easier than trying another AppImage builder.

dellus, 2024-05-25:
Sorry, it still is only 36 MiB and it doesn't run.

gerardbm, 2024-05-25:
I use this command: 

```
./LibreCAD-2.2.2_alpha1-40-g935acf5b.AppImage --appimage-extract-and-run
```

This is the output:

```
/tmp/appimage_extracted_f7b6336d45ef318d31f8053891913f80/usr/bin/librecad: error while loading shared libraries: libQt6Svg.so.6: cannot open shared object file: No such file or directory
```

dxli, 2024-05-25:
I cannot reproduce this error, and libQt6Svg.so.6 is included.

Could you try to find which libraries are missing for  libQt6Svg.so.6:

`./LibreCAD-2.2.2_alpha1-40-g935acf5b.AppImage --appimage-mount
`
then, run "ldd" for libQt6Svg.so.6

gerardbm, 2024-05-25:
Yes, it returns: `ldd: ./libQt6Svg.so.6: No such file or directory`

dongxuli-concord-design, 2024-05-25:
./LibreCAD-2.2.2_alpha1-40-g935acf5b.AppImage --appimage-mount

Will mount the filesystem, then, you can find the library file under its use/lib/ of the mounted

gerardbm, 2024-05-26:
I'm trying it today with the version alpha1-51 and it still says «No such file or directory».

I hope someone else can test it, as well.

dellus, 2024-05-26:
alpha1-51 doesn't run for me either.
At the download page it is stated as 46.2 MB, but when downloaded it is only 38.7 MB.
Maybe something wrong with the download mechanism?

dxli, 2024-05-26:
alpha-71 works for me now in a clean ubuntu-22.04 virtual machine.

dellus, 2024-05-26:
Wow, works for me now too! 
Version: 2.2.2_alpha1-70-g048a2684
Compiler: GNU GCC 9.4.0
Compiled on: May 26 2024
Qt Version: 6.6.3
Boost Version: 1.71.0
System: fossapup64 9.5

gerardbm, 2024-05-27:
I think it would run for me now if I had the GLIBC version updated (still 2.24).

I'm gonna make a virtual machine with Debian 12 to test it (between today and tomorrow).

Thanks a lot for the job you have been doing!

dellus, 2024-05-27:
alpha1-72 is a no go again for me on my fossapup64 9.5.
Tried then on bookwormpup, here it runs!
Is it the GLIBC issue gerardbm mentioned? What GLIBC version is needed? Fossapup has 2.31.
I have to say I am not delighted to be forced to use a newest system.

gerardbm, 2024-05-27:
> alpha1-72 is a no go again for me on my fossapup64 9.5. Tried then on bookwormpup, here it runs! Is it the GLIBC issue gerardbm mentioned? What GLIBC version is needed? Fossapup has 2.31. I have to say I am not delighted to be forced to use a newest system.

I have the version 2.24, it's old (Debian 9).

dxli, 2024-05-27:
We can pack glibc in. 

I suppose

gerardbm, 2024-05-27:
Tested from the virtual machine with Debian 12:

Versió: 2.2.2_alpha1-72-g6c1cf70c
Compilador: GNU GCC 11.4.0
Compilat el: May 26 2024
Versió de Qt: 6.6.3
Versió de Boost: 1.74.0
Sistema: Debian GNU/Linux 12 (bookworm)

dxli, 2024-05-30:
closing this one as fixed

u2fly, 2025-04-10:
> We can pack glibc in.

@dxli, this required to be able to use latest AppImage (glibc 2.38) with actual Debian stable (glibc 2.36).

<details><summary>Crash Log (2.2.2_alpha AppImage / Debian stable):</summary>

```
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by ./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by ./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libQt6Gui.so.6)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libQt6Gui.so.6)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libQt6Network.so.6)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libglib-2.0.so.0)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libxkbcommon.so.0)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libgssapi_krb5.so.2)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libicui18n.so.74)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libicui18n.so.74)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libicuuc.so.74)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libicuuc.so.74)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libdbus-1.so.3)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libkrb5.so.3)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libk5crypto.so.3)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libkrb5support.so.0)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libgomp.so.1)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libsystemd.so.0)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libgio-2.0.so.0)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libduktape.so.207)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libduktape.so.207)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libcap.so.2)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libgcrypt.so.20)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libgnutls.so.30)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libldap.so.2)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/liblber.so.2)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libmount.so.1)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libselinux.so.1)
[... cut ...]
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libtasn1.so.6)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libsasl2.so.2)
./LibreCAD-2.2.2_alpha1-382-gc3f1d8339.AppImage: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_LibreCeaDdgE/usr/bin/../lib/libblkid.so.1)
```

</details>

dxli, 2025-04-10:
I will try again.

u2fly, 2025-04-10:
> can you use snap or flatpak

Snap - no.

Flatpak - yes, but are there continious builds for 2.2.2_alpha in Flatpak format?

dxli, 2025-04-10:
@u2fly 

Currently, I have trouble with appimage for qpa plugin support. Issue #2100

u2fly, 2025-04-10:
@dxli 

To note, 2.2.2_alpha AppImage builds built _before CI switch to Ubuntu 24.04_ worked fine on Debian stable.

dxli, 2025-04-10:
the ci switching was because GitHub discontinued Ubuntu-20.04, the only version with a proper glibc version.

I couldn't get it built on ubuntu-22.04 due to qmake crashes.

That was the switching to 24.04. several things together: end of Ubuntu 20.04, crashing 22.04, GitHub Qt installation failure(fixed by GitHub now).

u2fly, 2025-04-17:
> Ubuntu-20.04, the only version with a proper glibc version.

@dxli 

Is it still possible to built latest LibreCAD 2.2.2.x on _local machine_ with Ubuntu-20.04?

Instruction outdated and does not inlude actual minimum reqirements for LibreCAD 2.2.2.x
- https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source#debian-ubuntu-and-other-derivatives