LibreCAD/LibreCAD #1792: Error: Assertion failed! Library Runtime C++ VisualMicrosoft

NabilAKROUD, 2024-05-28:
I have a large dxf file and every time I open it I get this error. 
I don't know the origin of the error.
To solve this problem I have been deleting elements until I found an entity that was giving me the error.
I leave the file with the suspicious entity [here](https://github.com/LibreCAD/LibreCAD/files/15466986/error.zip) in case someone finds it useful.
I have tried to open it with the x64 version, LibreCAD closes before opening it.

> Assertion failed!
> 
> Program: C:\Program Files (x86)\LibreCAD\LibreCAD.exe
> File: lib/engine/lc_looputils.cpp
> Line: 307
> 
> Expression: std :: get<bool>(current)
> 
> For information on how your program can cause an assertion
> failure, see the Visual C++ documentation on asserts
> 
> (Press Retry to debug the application - JIT must be enabled)

[image]

dxli, 2024-05-29:
Your hatch edge contains some invalid edges (zero length lines).

Fixed exceptions in hatch area calculation.

dxli, 2024-05-29:
fixed by 554f21374

closed 2024-05-29 by dxli