LibreCAD/LibreCAD #1793: Appimage v2.2.1_rc2 fails to launch

agambier-iem, 2024-05-30:
### Expected behavior

App launch when running app image.

### Observed behavior

Fails to start with error message : 
```
└> $ ./LibreCAD-2.2.1_rc2-6-g0790f21b-x86_64.AppImage 

(AppImageLauncher:7659): GLib-CRITICAL **: 09:21:48.534: g_file_test: assertion 'filename != NULL' failed
execv error: No such file or directory
```

### Steps to reproduce or sample file

- Download app image [v2.2.1_rc2.appimage](https://github.com/LibreCAD/LibreCAD/releases/download/2.2.1_rc-latest/LibreCAD-2.2.1_rc2-6-g0790f21b-x86_64.AppImage)
- make it executable
- run it

### Operating System and LibreCAD version info

```
└> $ lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 21.3
Release:	21.3
Codename:	virginia

└> $ uname -a
Linux RD-AGA 6.5.0-1023-oem #24-Ubuntu SMP PREEMPT_DYNAMIC Tue May  7 14:26:31 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

```

dxli, 2024-05-30:
32434db
Switched from appimagetool to linuxdeploy

dellus, 2024-05-30:
2.2.1_rc2_6 runs for me.
Version: 2.2.1_rc2-6-g0790f21b
Compiler: GNU GCC 11.4.0
Compiled on: May 27 2024
Qt Version: 5.15.3
Boost Version: 1.74.0
System: fossapup64 9.5

Version: 2.2.1_rc2-7 does not.

agambier-iem, 2024-05-31:
Same issue with latest 2.2.1_rc2-7

```
└> $ ./LibreCAD-2.2.1_rc2-7-g32434db1-x86_64.AppImage 

(AppImageLauncher:5243): GLib-CRITICAL **: 09:22:42.547: g_file_test: assertion 'filename != NULL' failed
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 483, resource id: 8501209, major code: 40 (TranslateCoords), minor code: 0
execv error: No such file or directory
```

dellus, 2024-06-01:
2.2.1_rc2_8 runs for me now.

Version: 2.2.1_rc2-8-g0649fad1
Compiler: GNU GCC 9.4.0
Compiled on: May 31 2024
Qt Version: 5.15.2
Boost Version: 1.71.0
System: fossapup64 9.5

dxli, 2024-06-05:
@agambier-iem what's the error message? 

And  your system?

noanight, 2024-06-05:
fails too....

user@host:~/Applications$ ./LibreCAD-2.2.1_rc2-8-g0649fad1-x86_64_53e9c1c6da66324d70ba73d804791d59.AppImage 
execv error: No such file or directory

user@host:~/Applications$ ./LibreCAD-2.2.0.2-14-g10ad9444-x86_64_22d2293bc471b1438e8bb41cfd370197.AppImage 
execv error: No such file or directory

user@host:~/Applications$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy

dxli, 2024-06-06:
This appears to be known issue of AppImageLauncher, as reported by 

https://github.com/TheAssassin/AppImageLauncher/issues/559

The solution so far is to uninstall AppImageLauncher.

Should we close this one?

agambier-iem, 2024-06-07:
Indeed after uninstalling appimagelauncher it wokrs !

This issue can be close since not related to LibreOffice image itself.

Thanks for the support

closed 2024-06-07 by dxli