LibreCAD/LibreCAD #1794: Strange window appears when program is closed when using QSS

Questions2000, 2024-05-31:
LibreCAD 2.2.1_alpha-434-g16afc25f
Appimage
Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8

I noticed that if I link a QSS (QT Style Sheet) file in "Options - Widget Options"
when I close LibreCAD a different window opens up.
It appears to be just the drawing space of LibreCAD that takes up the entire window.  The menu bar, toolbars, status bar, etc. all not shown.  The only thing this window shows is window header horizontal/vertical scrollbars.

Not sure if anyone else would be willing to verify this.
Does anyone know if this is a bug? or does anyone know how to prevent this from happening?

Thank You

Questions2000, 2024-06-05:
A little more information to share about this topic if anyone is interested.
I was just trying to test out a basic QSS file just to see if it would work.

I created a text file named `test.qss`
Inside this text file I added the following and then saved the file
```
QMenu { font-size: 24px; }
QTabBar { font-size: 24pt; }
QToolBar { icon-size: 32px; }
```

When you select “Widget Options” from the ”Options" drop down in menu bar a dialog box opens.
If you click on the button that says “Style Sheet” and navigate to my “test.qss” file mentioned above this applies the style sheet.

The two font based items work fine, the size of the fonts increase for context menus & tab labels.
The QToolBar icon line, does not work, but this does not work in any program I have tried it in, so not sure why this item never works in any program.  (I know LibreCAD provides an icon size option already but just was trying to get this line to work in a QSS file)

Anyways, I noticed that when the “Widget Options” dialog contains "path/test.qss" any time you go to close LibreCAD, that is when that extra window opens that I mention above in my original bug report.  Also noticed that this occurs whether the “Style” box is checked or not in the “Widgets Options” dialog.

--------------------------------------------------------------------------------------------------------------------------------
I also tried clearing out the QSS field information from “Widgets Options” and added the path to the launch command itself, but the same issue occurs.
I did this by creating bash script that contained the following
path-to-appimage/LibreCAD-2.2.1_alpha-434-g16afc25f-x86_64.AppImage -stylesheet path-to-QSS-file/test.qss

--------------------------------------------------------------------------------------------------------------------------------

Not sure if anyone can confirm this.  I am assuming that this is a bug.
If anyone tries this, just remember sometimes you have to save the "Widget Options" and close LC completely and then relaunch for the changes to take effect.

Thank you to anyone who reads and willing to try this out

dxli, 2024-06-06:
@Questions2000 

Clearly, you are the expert here.

Is this appimage only?

or you can reproduce this from a local debug build, hopefully, running as "librecad -d6"

Questions2000, 2024-06-09:
Oh no, I am no expert.
To answer your question, I am currently only using the generated Appimage version.  
My repo does not offer LC2 and I have never tried compiling it.
Using LibreCAD-2.2.1_alpha-434-g16afc25f-x86_64.AppImage

If anyone is willing to try.
`STEPS TO REPRODUCE:`
-Create a text file called test.qss
-Paste the following into the text file and save
`QMenu { font-size: 24px; }`
-While in LC click on “Options - Widget Options” in menu bar
-Click “Style Sheet” button and navigate to that test.qss file and then hit the “Save” button in dialog
-Then close LC
When you do that is when the blank window loads up.  

`STEPS TO STOP THIS WINDOW FROM SHOWING UP`
-Go back into “Widget Options” and remove the link to the test.qss file and hit save button again.
-When you close LC that blank window will load one more time.  But will not be there when you launch your next session of LC.

If anyone else can reproduce, please mention it.
Also if anyone knows what in the code may be causing this to occur.

There is a section about being able to use QSS in the LC manual, but this “bug” kind of prevents this from being used because users probably don't want to have to close this extra window each time they close LC.

Thank You

Questions2000, 2024-06-27:
I also tried this out using older version of LC and experienced same results
LibreCAD-2.2.0.2-14-g10ad9444-x86_64.AppImage

In reading a little of issue 1810 and posts linked to it, I tried/noticed something else.
If you close by clicking on the drawings “X” then the drawing closes and LC stays open.
If I then click on LC program close “X” then it closes out LC and that blank window I mentioned DOES NOT show up.

Seems like if you close out using LC  program close “X” or use your DE hotkey assigned (often Alt+F4) that LC should close all the drawings within (which is what it does when no QSS (style sheet) path is assigned), so still seems like a bug.

Not sure if anyone can confirm?
Thank You

Questions2000, 2025-01-03:
Created a new post to try to get all the information about this issue into one single post.  I have linked this post in that new post.

closed 2025-01-03 by Questions2000