LibreCAD/LibreCAD #1795: Hide the relative-zero marker still shows up with each launch

Questions2000, 2024-05-31:
LibreCAD  2.2.1_alpha-434-g16afc25f
Appimage
Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8

When you check the box for the following setting
"Preferences - Appearance - Hide the relative-zero marker"
it appears that this marker still shows up with each launch of LibreCAD, even though the setting remains checked on.

If you then open up the preferences window and just click the OK button, the action of this then causes this marker to hide as expected.  As mentioned though this happens with every launch of LibreCAD.

Not sure if anyone can reproduce the same result?

Hope this is helpful
Thank You

dxli, 2024-05-31:
fixed

Questions2000, 2024-06-05:
@dxli 
Wow, thank you very much for fixing this so quickly.

closed 2024-06-05 by Questions2000