LibreCAD/LibreCAD #1796: [ENHANCEMENT] Ability to turn off case sensitivity in command line

Questions2000, 2024-05-31:
Not sure if anyone else has experienced this but often when I type text in LibreCAD I use all uppercase letters, so I have my Caps Lock key ON, and then I go to type in a command and the command line does not recognize the command because the letters are capitalized.
I am not aware if this would conflicts with anything in LibreCAD, but I was wondering if it would be considered to add an Option to make the command line not be case sensitive.
I have seen other programs in Linux that allow different hotkeys to be assigned to both upper and lowercase letters, but I am not sure if LibreCAD is one of them.
Anyways was just wondering if anyone would find something like this useful / if this was even possible.

Thank You

dxli, 2024-05-31:
The easiest is to do it at with your local librecad.alias file:

https://wiki.librecad.org/index.php/Commands#Command_Alias_File

I won't recommend to implement this as a feature:

1. To keep the code base clean;
2. The feature is not universal, for example, not all languages have the concept of upper/lower cases;
3. Workaround works better here, for example, using a librecad.alias file to combine upper/lower cases.

Questions2000, 2024-06-09:
@dxli
Thank you for your response.
I had thought about using the custom hotkey file before making the post.  Am I correct in saying that if this was to be achieved using the custom hotkey file that I would have to create a copy of every tool again and replace lowercase letters with upper case letters?  Or is there an easier methoed that I am unaware of?

Thank You

Questions2000, 2024-06-22:
@dxli 
@Anyone
Before I close this post out, it is okay if it is (although will be a little time consuming to double each command), but can someone confirm that this is what is being suggested?

Example:
Adding the following to “librecad.alias” file
l	line
L	line
t	trim
T	trim
etc.

This post actually lead to many discoveries, I have made a couple of new posts because of it, some bug reports and some enhancement requests.  One of them was the realization that "keycodes" do not support uppercase, so if I was to want to use keycodes, I would not be able to use the solution mentioned.  I made a separate request about this here #1807 if anyone is interested.

Thank you for any information shared

Questions2000, 2024-10-21:
@sand1024 
With your "issues to close" post your goal was to close some issues which I have been trying to help out with as much as I can.  No rush but I was wondering if you can read this request post above and decide if this would be something worth adding in the future, if not I can close this.
Thank You

closed 2024-10-21 by Questions2000