LibreCAD/LibreCAD #1801: Lua Support

Questions2000, 2024-06-05:
Sorry, just a quick question.
I have been informed that LibreCAD 3 is no longer in development.
One of the planned updates for LibreCAD 3 was going to be the support for Lua scripting for tools/plugins.
I was wondering if there are any plans to now add support for Lua in LibreCAD 2 instead?

Thank You

dongxuli-concord-design, 2024-06-05:
I suppose we can do python instead, but have no idea on how.

lordofbikes, 2024-06-06:
For quick questions and discussions we prefer the forum https://forum.librecad.org.

In https://github.com/LibreCAD/LibreCAD_3/issues/403#issuecomment-2143353297 I just said, that **currently** nobody is working on V3.

V3 already has LUA, because everything in V3 started with LUA.
To implement a modern scripting language into the mature V2 code base was checked more than once and mostly discarded because of the core concept in V2 not being easy adaptable to available libraries.

closed 2024-06-06 by lordofbikes