LibreCAD/LibreCAD #1802: Colors on icons in the toolbar very hard to see

bit4man, 2024-06-06:
This is vanilla launch of LibreCad (screendump)

[image]

Perhaps it's due to me being color blind, but it's extremely hard for me to see what these icons do. 

### Expected behavior
Light on dark background cons. Or dark on white icons.  Not very dark on "almost not so dark" icons.

### Observed behavior
See screendump.

### Steps to reproduce or sample file
I've looked but been unable to find a place to adjust the icon colors.

### Operating System and LibreCAD version info
Version: 2.2.0.2
Compiler: GNU GCC 14.0.1
Compiled on: Jan 25 2024
Qt Version: 5.15.14
Boost Version: 1.83.0

dxli, 2024-06-06:
I feel the issue exists in Qt6 builds as well. Highlighting is hard to notice for push buttons.

lordofbikes, 2024-06-06:
closing as duplicate of #1144

closed 2024-06-06 by lordofbikes