LibreCAD/LibreCAD #1803: Cannot immediately reload command file

larrywesterman, 2024-06-06:
### Expected behavior
Immediately reopen command file to be executed in command window, using drop-down menu and selecting 'Load command file'

### Observed behavior
Previously opened command file is not available for reload

### Steps to reproduce or sample file
In drop-down command file menu, select 'Load command file'. Navigate to folder containing file. Select file and Open. Commands execute. Again select 'Load command file'. Previously loaded file is grayed out and cannot be selected. Select Cancel. Again select 'Load command file'. Previously loaded file is now black and can be selected. 

### Operating System and LibreCAD version info
Mac Sonoma 14.5, LibreCAD 2.2.0.2

larrywesterman, 2024-06-06:
Note that performing other actions between first and second load attempt does not change this behavior. Only attempting to reload the file and cancelling will allow second load.

dxli, 2024-06-07:
Fixed in master