LibreCAD/LibreCAD #1805: Command line widget height

Questions2000, 2024-06-09:
LibreCAD  2.2.1_alpha-434-g16afc25f
Appimage
Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8

Not sure why but with using the newest Appimage available it appears that the Command Line widget now has a minimum height that is rather tall.  There does not seem to be a way to adjust its height to only show a couple of lines.  The drag grip / widget height stops adjusting at this tall height.
This happens both when docked or floating.

Was wondering if anyone else was experiencing this?
Is this a bug?

Thank You

dxli, 2024-06-11:
@sand1024 , can you have a look? Probably, together with the pen widget placement. 

Regards!

sand1024, 2024-06-12:
yes, but from the very first glance, I can't see this on my environment, will check it with fresh install.  However, I need some time for this issue, I have to to finalize the current scope I'm working on first... so probably will not be able to fix it faster than in several days. Yet definitely will check

Questions2000, 2024-06-22:
@dxli
@sand1024 
Thank you guys for the responses.

Not sure if this helps, but I tried to look back to find a Linux version of LC that does not contain this height limitation.  The following versions height works well
LibreCAD-2.2.0.2-14-g10ad9444-x86_64.AppImage
it appears that newer versions that came out after this version are the ones that contain this height issue.
Hopefully this helps

Thank you again

ksio89, 2024-06-25:
@Questions2000 Thank you for posting this, the Cmd widget minimum height on 2.2.1 builds is indeed still excessive. Also had to downgrade to build 2.2.0.2, even if it's not a native 64-bit application like version 2.2.1.

dongxuli-concord-design, 2024-06-26:
@sand1024 , I am wondering whether we may move the command widget more flexible with its location and size

sand1024, 2024-07-01:
Hi guys, 

sorry for the delay with reply, I was in trip and had no chance to respond/fix earlier. I've made initial changes for command line widget width as well as some additional cleanup already. 

However, it seems it is necessary to make some additional testing and extract these changes from my working branch to separate one  - I think I'll create pull request one of these days.  

In general, I suspect this issue with height is related to qt6 - I've noticed some suspicious docking behavior with left toolbar too, yet need to investigate this deeper. 

Anyway, should commit soon. 

@dxli - Dongxu Li,  could you please also expand on "Probably, together with the pen widget placement." - do we have any GitHub issue  for this? I'm a bit confused  - probably I've missed or forget something there... 

Thank you

dxli, 2024-07-10:
@sand1024 

Hi, sorry for the late reply. 

I feel the right docking area is getting crowded, just wondering whether we could organize them, including the pen pellet widget better. 

Regards!

sand1024, 2024-07-10:
Hi!
Well, I suppose this issue is rather related to specific screen size.  For small screen resolutions - yep, I suppose it might be that there will be not not enough space on the screen, so tabs are the only option for placing widgets (or disable them via "Dock Widgets" menu items. 
  
For larger screens - the user may doc each widget in appropriate area - say, at the bottom toolbar of the screen.  
Something like this (sorry if the the image is too large) - 
[image]

So basically, I'm not sure that we can do something there... Of course, if you have any ideas how that might be improved - please let me know, yet I think we may leave this to the user-specific configuration...

Questions2000, 2024-07-13:
Sorry to add to this but here are just some opinions/ideas about the command line widget if anyone is interested.

`COMMAND LINE WIDGET`
-Like the ability to place this widget anywhere.  I personally dock this to bottom of screen.

-As mentioned already, older versions of LC allow the CLW height to be all the way down to just showing the CL field which I like the flexibility of this.  I like to show 3-4 lines personally.

-Header Bar
Not sure if anyone else would agree, but I would say if possible when CLW is docked to not show the vertical header bar shown on the left side.
This shows the header title "Cmd" which shows to the left of "Command:" text that is above the field itself, so this is shown already.
Also not sure how often users would need the "float" icon and also the "x" close icon.
There is icon already to show/hide bottom dock and the "float" option can be found in the icon drop down located in right side of command line field.
If you guys or other users do want them, might be a little strange not sure / also might not be possible but maybe could put in the blank space located all the way to the right of where it says "Command:" so that that acts like the header bar.
Personally I would not show cmd / float icon / x icon at all but not up to me.

Curious if anyone would agree to any of this above.
Thank you guys again working on this project, really great.

Questions2000, 2024-07-20:
A little more to add if anyone is interested.
I look at the Command Line widget as slightly unique compared to the other widgets based on where it gets docked.
Maybe can have two slightly different versions maybe using an “if” statement.
One for when docked to top or bottom.
One for when docked to right or left.

`Top or Bottom Dock`
When used in these docks maybe can be slightly different.
Basically ideas I already mentioned in my replies above this post.

`Right or Left Dock`
As mentioned I do not dock my CL widget to side docks personally, but I noticed that if you do, then you get a long vertical header on its left side.  Maybe when docked to right or left can make header at the top of the widget to match all the other widgets.

sand1024, 2024-07-20:
well, you're right regarding the specifics of Command widget. 

Yet actually, I feel that in general the entire concept of docking might be reviewed and refreshed.  For example, positioning or title based on doc area (as you've mentioned), potentially - autohiding (say, if left bar with action toolbars is used), separate view for print preview mode. 

Hope I'll take a look to this later and improve, yet can't say that this is the highest priority right now... However, with time I think we'll improve this.

Questions2000, 2024-07-23:
@sand1024 
Thank you for reading/responding to my suggestions.  Yeah no problem I was just mentioning some ideas because other post was now merged with mine.  Yeah you would know when the best time is for those docking updates to occur.
Hopefully fixing the CL height adjustments to match what they were in previous LC versions mentioned as the main topic of this post will occur before them.  Not sure if this will have to wait to be handled along side the docking updates in the future, again you would know more than I would.
Just thank you in general for working on the project, really appreciate it.

Questions2000, 2024-08-08:
I was not able to test this because newer LC AppImages did not open.
I was finally able to make some progress with AppImages and was able to get newer versions to open and was able to test out the Command Line Widget height, and I noticed that this was now fixed.
Just wanted to say thank you for working on this, the quick test I did appears to work great.
I will keep this post open because it was merged to contain other widget improvements to be looked at in the future.
Thanks Again

Questions2000, 2024-09-29:
@sand1024 
The initial issue of this post has been fixed (thank you again for this)
but there was more information discussed about the command line widget.
Should I create a new post containing any information this post contains that was not addressed?
Thank You

sand1024, 2024-09-29:
Yes, it's always better to create separate and focused issues rather ones that covers several concerns as part of single ticked.

Questions2000, 2024-10-02:
@sand1024 
Okay, I just did because my post was originally merged by another member, so I thought you guys wanted all command line content in same post.  I agree, as per your request I will separate this post out into separate posts so easier to understand and keep track of.
Just letting you know that I have closed this post if you want to take it off your "Issues to close" list.
Thank you

closed 2024-10-02 by Questions2000