LibreCAD/LibreCAD #1806: RS_Text: the width factor setting may cause incorrect font shapes.

dongxuli-concord-design, 2024-06-19:
RS_Text allows specifying a width factor to scale the fonts in the horizontal direction. If the width factor is not 1, and the font contains an arc in the font polyline, the scaling will result in an incorrect font shape.

https://youtu.be/oGRxh51QI9s

### Expected behavior:

Setting a width factor will scale the text in the horizontal direction.

### Observed behavior:
Distorted font shapes for font arcs.

### Steps to reproduce:
1. Menu: Tools > Dimensions > Text;
2. In the Text dialog, choose:
3. Font: Standard
4. Width Factor: 2
5. Text: B
6. Okay to insert the text.

The actual result: the letter B is rendered incorrectly.

dongxuli-concord-design, 2024-06-19:
The root cause:

LFF fonts are implemented as polylines, which may contain line segments and arcs.

If the scaling factor is different in xy-directions, arcs are scaled to elliptic arcs, so an arc containing polyline cannot be scaled properly while staying as a polyline.

Possible solution:

When scaling anisotropically, _i.e._ differently in xy-directions, explode the polyline and scale the entities separately.

app4soft, 2024-09-01:
> Possible solution:
>
> When scaling anisotropically, i.e. differently in xy-directions, explode the polyline and scale the entities separately.

Looks OK as a workaround.

Also, could this method be used in LibreCAD to create _isometric projection_ of selected orthogonal view sketch?

It looks similar to how QCAD's "Isometric Projection" tool works:

- https://qcad.io/rsforum/viewtopic.php?t=8795
  - https://www.qcad.org/doc/qcad/latest/reference/en/scripts/Projection/IsometricProjection/IsoProject/doc/IsoProject_en.html
  - https://www.qcad.org/doc/qcad/3.7.0/reference/en/scripts/Pro/Projection/IsometricProjection/IsoProjectCylindrical/doc/IsoProjectCylindrical_en.html
- https://www.youtube.com/embed/_L5H4nTFxx0