LibreCAD/LibreCAD #1808: Keycode command line text

Questions2000, 2024-06-22:
LibreCAD AppImage
Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8

Couple requests regarding keycodes text that shows up in command line.
1.) When using keycodes mode, when you type in a command it shows the following item in the command line
"Accepted Keycode:"
was wondering if it could be considered to change this to the following instead
"Keycode:"
This is makes each item a simpler name, similar to how LC uses "Command:"
Also all keycodes are accepted, because if they are not, the command just does not get activated.

2.) When using keycodes, it also does not give any indication of the command being run, like how regular "Command:" does
Maybe can add the command similar to how regular commands do?
Example
Command: l (line)

Thank You

Questions2000, 2024-08-27:
@dxli 
Thank you for working on this, everything works great but I just noticed one minor thing if interested.
For commands they show up in the command line like the following
`"Command:"`
`"Unknown command:"`
but keycodes show up like
`"keycode:"`
`"unknown keycode:"`
Would it be possible to capitalize so looks like
`"Keycode:"`
`"Unknown keycode:"`
to match the "Command" items

I prefer titlecase personally but you may not, so can just go with what is mentioned above if decide to work on this.
Titlecase (sometimes called Propercase)
`"Unknown Command:"`
`"Unknown Keycode:"`
Anyways, thank you again for the updates made.

Questions2000, 2024-08-27:
Sometimes I prefer titlecase personally but you may not, so can just go with what is mentioned above if decide to work on this.
But this is what Titlecase (sometimes called Propercase) would look like where both words are capitalized
`"Unknown Command:"`
`"Invalid Keycode:"` 
or if decide to change
`"Unknown Keycode:"`

Anyways, thank you again for the updates you have made to this so far.

Questions2000, 2024-08-31:
(If this title case is not agreed on then can just ignore this reply)
If the "title case" idea is something that developers agree with, it could make those messages standout a little bit more visually, from the content that appears after the ":"
such as commands, Linux paths, etc. which are typically mostly lower case text.

Anyways if this is decided upon, I just realized that there might be other multi word phrases that would need to be updated as well.
I am not aware of them all, but here are another two for example that I recently realized

When opening a drawing
“Loaded document:”

When saving a drawing
"Saved drawing:"

dxli, 2024-08-31:
I would agree being consistent is better.

I suppose this is an issue of English messages. Not all languages share the same set of capitalization rules, or even have alphabet.

Questions2000, 2024-08-31:
@dxli 
Thank you for responding.
So does this mean that something like this will be added?
At least in the English language version?
Not sure how languages are handled.

If not would it be possible for someone to at least capitalize "Keycode:"
"Invalid keycode:"
just to match the two "Command:" items.

Thank you again

Questions2000, 2024-10-02:
@dxli 
@sand1024 

This one was being worked on but was stopped.
It says Fixed, but not fully complete.
This would just require capitalizing some words in the command line if anyone would like to finish this and then could be closed in regards to "Issues to close" 1902 list.

closed 2024-10-21 by Questions2000