LibreCAD/LibreCAD #1809: librecad.alias does not seem to work with newer AppImage versions

Questions2000, 2024-06-22:
LibreCAD-2.2.2_alpha1-98-ga3bb1b64.AppImage
Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8

I noticed that the custom hotkey "librecad.alias" file does not seem to work in newer AppImage versions.
I went back and tried an older AppImage version and it does.
The following version works
LibreCAD-2.2.0.2-14-g10ad9444-x86_64.AppImage
but newer versions do not

Thank You

dongxuli-concord-design, 2024-07-06:
@Questions2000 

can you get the debugging messages?

1. LibreCAD-2.2.2*.AppImage --appimage-mount
2. cd to the appimage mount folder:
3. usr/bin/librecad -d6 |grep 'alias'

Questions2000, 2024-07-10:
@dxli
Thank you for looking into this.
I will if I could, but I was not able to get this to work.
I tried just launching AppImage and then going into the tmp folder where it gets mounted.
I was able to find path you mentioned usr/bin/librecad
The "librecad" is an exec file so I was not able to search it like you mention
I apologize I do not know what -d6 does
I tried `librecad -d6 |grep 'alias'` from that path.
Let me know if there is anything else I can try.

Questions2000, 2024-08-12:
@dxli 
Okay, turns out the other issue I was having where LC would close instantly right after splash screen (1817) was due to something with the librecad.alias file.
I had an alias file from my previous instances of LC where I just added a couple of test hotkeys to the blank alias file that gets provided by LC.
I deleted this file and all of a sudden new AppImage versions of LC now all open fine.
After running new version LC, I then saw that LC created a new alias file in the same location.  I opened this alias file and noticed that now this alias file is no longer a blank file, but now contains all the commands typed out, which is nice.
However, just for a test I tried to just add a new custom hotkey to this file, and after doing so I then noticed that the issue of closing LC right after splash screen occurred again.
So the closing issue appears to be when new custom items are added to the alias file.
Not sure if anyone could verify this / know what would be causing this issue.
Hopefully this is helpful
Thank you again for all your help

dxli, 2024-08-12:
I will see why a crash could happen due to the alias file

Questions2000, 2024-08-27:
I don't know if you had time to look into this yet, but I was just wondering if you did, did you have any luck narrowing down why the alias file would be causing this issue?
As mentioned the alias file used to be blank but now contains a full list of commands typed out so it was worked on by someone.
I played around with this alias file, seems that no matter what is added to this file, if the file is in any way different then its default state, it prevents LC from opening.  Even if you just try modifying one of the existing entries with two capital letters instead of lowercase the issue occurs.
Anyways if you need me to test anything just let me know.

Whoever did modify this alias file thank you for taking the time to type in all those commands, obviously does not work right now but when this is fixed this will be very helpful.

Also is there anyone else willing to verify this?  Does anyone else experience this issue when making modifications to the alias file when using the newest continous release AppImage version for Linux?

Thank You

Questions2000, 2024-08-31:
@dxli 
Not sure if it was you or another developer but I tried the newest AppImage available
LibreCAD-2.2.2_alpha1-150-gdc895a29.AppImage
and this now is fixed.
This bug was a concern of mine, thank you guys for this.

closed 2024-08-31 by Questions2000