LibreCAD/LibreCAD #1810: On exiting LibreCAD appImage, I now have to close 2 windows.

JohnCC330, 2024-06-23:
### Expected behavior

On Closing LibreCAD, on closing with the top-right 'X', all should close

### Observed behavior

On using 'X', LibreCAD closes partly, leaving another copy of the drawing, which then has to be closed too.

### Steps to reproduce or sample file

Just start the LibreCAD appImage, and hit 'X'. Another copy of the (empty) drawing is shown (minus the top
menubar). 

### Operating System and LibreCAD version info

Slackware64, 
Version: 2.2.1_rc2-11-g933d5378 (had the same issue with 2.2.0.1 version)
Compiler: GNU GCC 9.4.0
Compiled on: Jun 17 2024
Qt Version: 5.15.2
Boost Version: 1.71.0

Otherwise, all operations seem to run smoothly.

dellus, 2024-06-25:
Seems to be a Slackware related issue. I don't have a full Slackware installed but tested on a Puppylinux slack variant. In contrast to other OSes running LibreCAD there are two 'top-right X , the upper closing LibreCAD completely (quit), the lower closing the active drawing only (close).
If you are in Fullscreen mode (F11 or View - Fullscreen) the upper Menu or title bar is disappeared and thus, when clicking on the remaining 'X' you just close the drawing.
It seems Fullscreen mode is saved for the next time you start the appimage.

JohnCC330, 2024-06-25:
I'm not sure if I follow you. I doubt it's a Slackware X11 issue. Maybe it's a XFCE4 problem?
Earlier on, I compiled LibreCAD versions, and never noticed this problem (probably till version 0.21 or so). I did not have the time to search for a compilation issue on the latest LibreCAD version, so I decided to run the appImage. Since then I have the double-close issue.

lordofbikes, 2024-06-25:
Sounds indeed like a duplicate of #1557

dxli, 2024-06-26:
Doesn't seem to be a bug. 

If it's full screen, you need to quit full screen to see the window close button. 

Mistaking the document close button as the window close button is real, but the current behavior is the expected. 

For users using full screen, there are shortcuts to use.

Questions2000, 2024-06-27:
I apologize if this is not correct, but it sounds like the user is experiencing something similar to what I mentioned in my post #1794
A second window opens up that contains no toolbars / widgets / etc. just the drawing space panel.
Only difference is mine occurs when trying to use QSS.

JohnCC330, 2024-06-30:
> I apologize if this is not correct, but it sounds like the user is experiencing something similar to what I mentioned in my post #1794 A second window opens up that contains no toolbars / widgets / etc. just the drawing space panel. Only difference is mine occurs when trying to use QSS.

Exactly!

JohnCC330, 2024-07-12:
I didn't know what a QSS file is supposed to do - now I know a little more, and your insistence did lead me to identifying the cause (thanks!). 

Following instructions I read somewhere, I had a style sheet defined named 'darktheme', and if I clear that field, the double window indeed disappears. And if I redefine the style sheet, the second window disappears... Note that this is confusing: In both cases the 'style' checkbutton is disabled! (I guess the style checkbutton only applies to the drop-down - Windows/Fusion? )

So, I guess there is some issue with the style sheet. I located some docs on that, though I'm somewhat at a loss understanding why a stylesheet would/could  change operational aspects of a program...

dongxuli-concord-design, 2024-07-12:
The basic story of closing the app:

1 you may close all drawing windows, and the app will stay open. User can open a drawing or drag a dxf to the app window from a file browser;
2, user can close the app directly, but will be asked if any unsaved and modified drawing. 

How to trigger the closing scenarios is not really handled by the app, as it's the job of the os or window manager.

Questions2000, 2024-07-13:
Got bad feeling about this one

@JohnCC330
Okay, so that confirms that the issues we are experiencing are caused by the same thing.
Thank You

@dongxuli-concord-design 
I apologize if I am not understanding something, but I feel that it is important to mention again.  I do not want to speak for him, but I believe JohnCC330 & I our both using the 2nd “closing the app” method you mention above, closing out the entire app and all the drawing opened within it (doing so by clicking on windows close “x”, hitting Alt+F4, etc.)
Without the QSS file/path being linked in “Widget Options” LC closes perfectly as expected.  This only occurs when QSS path/file is linked in “Widget Options” field.  Seems like LC should close out completely the same way regardless of QSS or not.

So when you say....
"How to trigger the closing scenarios is not really handled by the app, as it's the job of the os or window manager."
I am a little confused, are you saying that the QSS file being linked is the thing that is changing the “closing scenario”?

I am not familiar with “Slackware” but odd that Slackware and Linux are both experiencing this so LC would be the common item in the situation.  I wonder is this occurs in Windows OS as well?

Also I have used QSS with other applications and no other programs experience anything like this double window issue either.

The manual has a whole section about QSS, so apparently this is something that was intended to be used in LC.  QSS allows you to do so much with the program.
https://docs.librecad.org/en/latest/ref/customize.html
"Style Sheets" at bottom of page

Anyways, just trying to help narrow down what could be going on with this issue.
Hopefully some of this is helpful / something that can be fixed.

Thank You

Questions2000, 2024-07-13:
I tried the QSS test I mentioned in my post #1794 adding QSS to the actual LC command itself (which also causes this issue)
Creating a test.qss file and also adding the following to a bash script
`path-to-appimage/LibreCAD-2.2.2_alpha1-98-ga3bb1b64.AppImage -stylesheet path-to-QSS-file/test.qss`

When launch LC this way, if you open up “Widget Options” the Style Sheet field is empty.  Also this dual window occurs whether the “Style” box is checked or not in "Widget Options"

If I can maybe I will try this on Windows OS to see if this occurs, if Windows uses QSS in the same way?

Questions2000, 2025-01-03:
@JohnCC330 
Glad I was able to help you a little bit.
Not sure if you are still around this site but I made a new post about this topic to try to get everything in one post to make it easier for the developers.
I will link this post in that new post as well.
Would you be willing to close this post, just for simplicity.
Thank You

closed 2025-09-29 by JohnCC330