LibreCAD/LibreCAD #1812: Modify/Customize librecad.alias

michele-bs, 2024-06-27:
**Expected behavior**
After rebooting LibreCAD, the keycodes that I have just edited on librecad.alias are recognized and work correctly.

**Observed behavior**
I change the `zoompan` keycode (`ActionZoomPan`) from `zp` to `pp`, then I write `#` at the beginning of the line 60 of _librecad.alias_ (`pp ptp`) to solve the conflict with `ActionDrawLineParallelThrough`, then I save _librecad.alias_ and reboot LibreCAD. If I write `pp` in the command line, `ptp` is still activated, instead of `zoompan`.

**Sample file**
[librecad.alias](https://github.com/user-attachments/files/16015751/librecad.alias.txt)

**Operating System and LibreCAD version info**
Win 7 Pro SP1
LibreCAD 2.2.0.2

dongxuli-concord-design, 2024-06-27:
Can you test with the latest development?

michele-bs, 2024-06-28:
I have just tried with LibreCAD 2.2.1; unfortunately, the issue is still here.

From the Cmd:
> pp
> Command: pp (polygoncencor)
> zp
> Command: zp (zoompan)

while `pp` should rather be `zoompan`, in my _librecad.alias_.

Any idea?

--
Version: 2.2.1_rc2-11-g933d5378
Compiler: GNU GCC 12.2.0
Compiled on: Jun 17 2024
Qt Version: 5.15.2
Boost Version: 1.75.0

dongxuli-concord-design, 2024-06-28:
The issue should have been fixed in the development branch 2.2.2_alpha

michele-bs, 2024-07-14:
I installed the 2.2.2_alpha, but unfortunately I can't even open LibreCAD, now. I see an error message regarding a dll (_api-ms-win-core-synch-l1-2-0.dll_), the same one [stated from this other Microsoft user](https://answers.microsoft.com/en-us/windows/forum/all/entry-point-not-found-api-ms-win-core-synch-l1-2/f978dfb0-f481-40ba-9022-63cde56e4f4c).

I have also tryed to reinstall [Visual C++ Runtime 2015](https://www.microsoft.com/en-us/download/details.aspx?id=48145) and 2017 and to reinstall [KB2999226](https://support.microsoft.com/en-us/help/2999226/update-for-universal-c-runtime-in-windows); and I also have tried to run System File checker from Command Prompt, with no success.

Can you suggest me a solution? Otherwise, I will have to downgrade to LibreCAD to 2.2.1_rc3, which doesn't incurr in this issue and works well, but also can't let me customise librecad.alias, as stated in the beginning of this thread.

Thank you.

> HI @michele-bs ,
> 
> I mean the master branch (2.2.2_alpha), for the latest.
> 
> Regards!

dongxuli-concord-design, 2024-07-14:
Just tested, the latest 2.2.2_alpha win64 build works for me on win11

Micha-he, 2024-08-08:
Maybe under Win11, on Win10 too. But not under Win7 x64 !

dxli, 2024-08-08:
Let's see whether we can build 2.2.1 on win7