LibreCAD/LibreCAD #1813: Line weight graphical toggle

Questions2000, 2024-06-27:
LibreCAD-2.2.2_alpha1-98-ga3bb1b64.AppImage
Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8

I apologize because I am a little confused, so this might be a bug report or a feature request if it does not exist.
I notice when using newer AppImage versions that now when you zoom in, it appears that lines, text, etc. show up as thick bold items.  In older versions such as LibreCAD-2.2.0.2-14-g10ad9444-x86_64 this does not occur.

I am not sure if this is on purpose that now it does, was this added as a feature? or is this maybe a bug?

While on the subject, if this is not a current feature, I would like to make a request if anyone else is interested, to offer a way to toggle graphical line weights shown on your screen on/off.
So if lines are thin they show up thin, but if you have line weights that are thicker in value, they show up on your screen as bolder lines.  User would be able to toggle this feature on/off because not always desired.

Thank You

dongxuli-concord-design, 2024-06-27:
It was requested that we make the default line width above 0(which means 1 pixel).

The reason is that new users can easily got confused with 0 width. It shows on screen, but hardly visible in printout.

dongxuli-concord-design, 2024-06-28:
We only tweaked the default width for layer 0, not for user created layers. 

It's better to generate a warning, if width zero or draft mode is requested for printing.

Enforcing a printing time line width is quite bad for drawing time.

Questions2000, 2024-07-13:
Oh okay I overlooked that 0 layer had a line weight assigned by default.
Okay so that is what makes its line weight show up when you zoom into anything on 0 layer.
That being said, is there any way of being able to turn this off visually?
So that no matter how much you zoom in all lines appear as thin lines?
If not, could someone add a "Feature Request" label to this post as well?
Requesting a toggle to be able to show/hide line weights graphically on screen.

Thank You

sand1024, 2024-09-16:
if you use LayerTree widget, there are settings dialog - and within this setting dialog it is possible to set default values for line that should be applied to newly created layer

Questions2000, 2024-09-29:
@sand1024 & Any Other Developers Interested
This was on "Issues to close" list, but it was also a feature request as well, but I do not think it has been added.

The feature request was for the following.
Add a toggle setting to show/hide line weights (widths) visually on screen.
-If OFF all lines visually would appear as thin lines
-If ON then lines would visually appear with different thicknesses on screen based on their line width values.
User might not want to have to see line thicknesses all the time, which I believe is currently the default.
The line weights become visible when you zoomed in.
Can be distracting/confusing to work zoomed in with line weights showing bold (if they contain heavier line weights)

If want to close this post specifically would it be alright if I created a feature request post again for this desired feature?
Let me know

Thank You

sand1024, 2024-09-29:
hm... I'm lost a little by your last comment, sorry.  Let me step back and summarize:

1) For the normal drawing mode, "Draft" mode controls how lines will be drawn (for draft - 1px regardless of line width, for "non-draft" - in screen width according to specified line width.  

Basically, the only difference of Draft mode from "non-draft" it is rendering of linewidths, simpler drawing for text, simpler drawing for hatches. 

2) In Print Preview - there is corresponding button that also switches line widths. 

As for default values for newly created entities - they are also supported. 

So, from my point of view - all cases are covered, I'm not sure I understand why separate setting for line width is needed.... 

Or you mean that you'd like to have such setting for Non-Draft mode only?  Please expand on.

Questions2000, 2024-10-03:
@sand1024 
Thank you for responding, as you mentioned better to keep things clean so I decided to close this post and create a new one, hopefully this is okay.
The issue number for this new post is 1912 which hopefully explains better, if still interested (intentionally not linked)
Thank you again

closed 2024-10-03 by Questions2000