LibreCAD/LibreCAD #1814: Have to type letter twice to get back into command field

Questions2000, 2024-06-27:
LibreCAD-2.2.2_alpha1-98-ga3bb1b64.AppImage
Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8

I noticed that when using the line tool in newer versions that there now is an extra step, where you have to type letters once to get the focus back into the command field and then again to start actually typing in the field

The following version contains this issue
LibreCAD-2.2.2_alpha1-98-ga3bb1b64

`Steps To Reproduce:`
-type “l” to activate line tool
-draw something
-RMB once to stop current line
-RMB again to stop the command itself
***This is where issue occurs, expect to just type in next command
-instead you have to type any letter which then brings focus to the command field
-then you can start typing your next command

In this older version this works as expected
LibreCAD-2.2.0.2-14-g10ad9444-x86_64

Not sure if anyone can confirm.

Thank You

michele-bs, 2024-06-30:
Yes, I have the same problem. Command field activates typing any key, and then it accepts instructions.

dongxuli-concord-design, 2024-07-04:
We will fix this one and tag 2.2.1_rc3

dongxuli-concord-design, 2024-07-06:
I have tagged 2.2.1_rc3 with this fix

michele-bs, 2024-07-14:
Fixed! Thank you. :-)

> I have tagged 2.2.1_rc3 with this fix

Questions2000, 2024-08-27:
Could not get AppImage to work for a while due to current alias file issue, but now at least can open the program to test this, agreed Fixed! Thank you as well for working on this works great.

closed 2024-08-27 by Questions2000