LibreCAD/LibreCAD #1817: New AppImage version does not open only shows splash screen

Questions2000, 2024-07-10:
Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8

I recently tried the latest continuous release AppImage
LibreCAD-2.2.2_alpha1-110-gbd7018ef.AppImage
and it only shows the LC splash screen when launched, it does not seem to open the application itself.
Not sure if anyone else experiences this?

Thank You

Questions2000, 2024-07-15:
Not sure if any of the following helps.
I tried a newer AppImage version of the continuous build just to see if this was just maybe a compiling issue that caused the last version not to work but still occurs.
Just shows the splash screen but the program does not load.
I also checked “task manager” to see if maybe the program was running in the background but it does not.
I also checked to see what happened in the “tmp” folder when AppImage was run.
It appears to open a temp mounting folder very quickly while the splash screen is showing but it disappears quickly from the “tmp” folder.

Questions2000, 2024-07-23:
Just trying to help narrow down what might be causing this.
The following was the last version I was able to get to work on my machine
LibreCAD-2.2.2_alpha1-98-ga3bb1b64.AppImage

This version is when I noticed this issue
LibreCAD-2.2.2_alpha1-110-gbd7018ef.AppImage

So what ever change that was made had to happen either to the version mentioned above or ones in between them.
Hopefully this is helpful

Does anyone know if older versions of the LibreCAD-2.2.2_alpha1 Continuous Builds AppImages are available anywhere to download?  Or do they just get overwritten with each build?

Just thinking maybe I could try versions in between the two versions I mention above to narrow down the exact version where this occurs, this maybe could help pin point the version / update that was made that causes this issue.

Thank You

Questions2000, 2024-08-08:
I have gathered some more information that I think could be helpful in figuring this out.
First off I discovered that these new AppImage versions do work fine.
I tested them out in multiple "live" instances of many different distros (GTK & QT based) and the AppImages open in all of them.
I tried a "live" version of the distro I was running when I ran into this issue of LC not opening and it even works fine with this.
So it is apparently something installed that is causing LC not to open.

I tried to help narrow this down and I started over and re-installed any QT based apps I had installed because I thought it might be some kind of QT based item that was causing this.
When doing this, the newer AppImages of LC worked fine after all my other QT application installs, so it might not be this.

Seems strange because I would think that LC AppImages contain everything they would need.
Does anyone know if LC AppImages require the use of any dependencies from your system itself?
Because as mentioned something installed seems to be causing the issue of LC not opening.

As mentioned that last version that opened fine on my system was
`LibreCAD-2.2.2_alpha1-98-ga3bb1b64.AppImage`
(Says this was compiled June 17th 2024 so not that long ago)
Versions I tried after this one, show LC splash screen and then closes automatically.
So versions after the one that work seem to be affected by something installed on my system.

I was finally able to get some more information from running LC AppImages from the Terminal.
I just copied the information that shows up at the end of terminal because these are the only parts that appear to be different.

`TERMINAL INFORMATION`

`LibreCAD-2.2.2_alpha1-98-ga3bb1b64.AppImage`
`(THIS IS ONE THAT WORKS)
```
font: /tmp/.mount_LibreCpLGBLO/usr/share/librecad/fonts/unicode.lff:
base: unicode
/tmp/.mount_LibreCpLGBLO/usr/share/librecad/patterns
/tmp/.mount_LibreCpLGBLO/usr/share/librecad/qm
QFSFileEngine::open: No file name specified
qt.core.qobject.connect: QObject::connect: No such signal LC_QuickInfoWidget::escape()
qt.core.qobject.connect: QObject::connect:  (sender name:   'LC_QuickInfoWidget')
QLayout: Attempting to add QLayout "" to QG_LibraryWidget "Library", which already has a layout
/tmp/.mount_LibreCpLGBLO/usr/share/librecad/library
/tmp/.mount_LibreCpLGBLO/usr/lib/librecad
0
```

I think the following is the newest one I tried
`LibreCAD-2.2.2_alpha1-121-g512680e6.AppImage`
`(THIS DOES NOT WORK)`
```
font: /tmp/.mount_LC-NEWnccJOA/usr/share/librecad/fonts/unicode.lff:
base: unicode
/tmp/.mount_LC-NEWnccJOA/usr/share/librecad/patterns
/tmp/.mount_LC-NEWnccJOA/usr/share/librecad/qm
QFSFileEngine::open: No file name specified
qt.core.qobject.connect: QObject::connect: No such signal LC_QuickInfoWidget::escape()
qt.core.qobject.connect: QObject::connect:  (sender name:   'LC_QuickInfoWidget')
QLayout: Attempting to add QLayout "" to QG_LibraryWidget "Library", which already has a layout
/tmp/.mount_LC-NEWnccJOA/usr/share/librecad/library
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
Aborted
```
(I renamed the AppImage to "LC-NEW" which is what the .mount_ path is showing above)

Hopefully this will be helpful in finding out what is causing this issue.

dongxuli-concord-design, 2024-08-08:
Can you mount the appimage, and run librecad with "librecad -d6"?

Need to find where the error occurs

Questions2000, 2024-08-12:
@dongxuli-concord-design 
Thank you for mentioning this, this time I was actually able to get -d6 to work and it turns out this has something to do with my other issue post about alias file.  I am going to close this post and explain what I found in my alias file issue post found here #1809

closed 2024-08-12 by Questions2000