LibreCAD/LibreCAD #1818: Fonts are monospaced only

hapees, 2024-07-12:
### Expected behavior
Display fonts as in previous versions.
If proportional fonts are selected keep them proportional. 
Like in:
Version: 2.2.0.2
Compiler: GNU GCC 14.0.1
Compiled on: Jan 25 2024
Qt Version: 5.15.14
Boost Version: 1.83.0
System: Fedora Linux 40 (Server Edition)

### Observed behavior
Current version allows mono spaced fonts only.
Proportional fonts in older drawings will be converted to mono spaced fonts when opened.

### Steps to reproduce or sample file
Enter Text and try to set proportional fonts.

### Operating System and LibreCAD version info
Version: 2.2.2_alpha1-112-g7b96f926
Compiler: GNU GCC 14.1.1
Compiled on: Jul 12 2024
Qt Version: 6.7.2
Boost Version: 1.83.0
System: Fedora Linux 40 (Server Edition)

dxli, 2024-07-14:
Could you be more specific? 

Including detailed steps to reproduce the issue? 

If I am not mistaking, our WenQuanYi fonts are of the same width for each character. For example, letters are narrower and Chinese characters are wider. 

Could you attach some screenshots to show the issue also?

hapees, 2024-07-15:
For Version: 2.2.0.2 the fonts are being rendered proportional,  as expected. 
[image]
 

For Version: 2.2.2_alpha1-112-g7b96f926 the rendering is mono spaced. No way to set rendering to proportional 
[image]

Please consider the alphabet characters can be mono or proportional spaced.

Having mono spaced text only would render older drawings more or less useless as updates with the latest LibreCAD Version  changes the font spacing for all the text in a drawing.

dxli, 2024-07-15:
@hapees 

Thanks a lot for the explanation, and I am clear on the root cause of this issue. 

In an attempt to fix #1786 , the text alignment algorithm is modified to use font width for alignment and that fix left all alphabet to be effectively misplaced. 

Their could be an easy fix those. The root cause is that I didn't understand the lff font design when I made the changes there.

hapees, 2024-07-27:
Hi,
I pulled the latest version.
In a quick check: the proportional spacing is OK.
Thank You

app4soft, 2024-08-23:
Probably related (or duplicate): 
- https://github.com/LibreCAD/LibreCAD/issues/1845