LibreCAD/LibreCAD #1819: Arc Length Dim doesn't recognize Dim'n General Scale

GSStnb, 2024-07-13:
### Expected behavior
Arc length text, arrow size, etc should be resized as determined by the Dimension "General Scale"

### Observed behavior
Arc length text, arrow size, etc is not resized.
 
### Steps to reproduce or sample file
Draw a square with an arc on one side
Dimension the square sides and arc length
Change the dimesion "General Scale", say x10 (Drawing Preferences - Dimensions).
[image]

### Operating System and LibreCAD version info
Version: 2.2.1_rc3-1-g4d19419b
Compiler: GNU GCC 9.4.0
Compiled on: Jul 11 2024
Qt Version: 5.15.2
Boost Version: 1.71.0
System: Debian GNU/Linux trixie/sid

dongxuli-concord-design, 2024-07-17:
looks like we need dxfrw support for arc length dimensions.

u2fly, 2025-04-16:
@dxli Arc Length still ignores changes in current document "Text style" (font). 

Version: 2.2.1.1
Compiler: GNU GCC 11.4.0
Compiled on: Mar 30 2025
Qt Version: 5.15.8
Boost Version: 1.74.0
System: Debian GNU/Linux 12 (bookworm)