LibreCAD/LibreCAD #1820: librecad 2.2.2-alpha based on qt6.6.3 global variables

ma12155506, 2024-07-22:
When I compiled and used librecad 2.2.2-alpha based on qt6.6.3, I found that the line width and text size of the global variables in dxf did not have an effect. Observing the code, it seems that some of the global variables were not read or used. The second nested editing block may cause the program to crash

[image]

dongxuli-concord-design, 2024-07-27:
could you show the call stack at the crash?

It also helps to find steps to reproduce the crash.

ma12155506, 2024-07-31:
[image]
I think I added the attribute 'attdef attribute' for text parsing, but there was an error when he tried to find non-existent blocks while editing the block. Now that it has been fixed, I'm not sure if this can be considered a bug。 
in librecad\src\ui\qg_graphicview.cpp
[image]

dongxuli-concord-design, 2024-07-31:
Could you test and send in a pull request?

ma12155506, 2024-08-01:
how  can  send in a pull request

dongxuli-concord-design, 2024-08-01:
The first step is to fork the project, then you can  clone, branch in your forked repository. 

Whenever you push to your repository, GitHub page will show you a button to create pull request under "Pull requests" tab in you repository page. 

https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request