LibreCAD/LibreCAD #1821: I need detailed documentation on developing LibreCAD plugins.

c61292558, 2024-08-05:
### Expected behavior
Hello, developers.
I need detailed documentation on developing LibreCAD plugins. The information on this website is too sketchy: [LibreCAD Plugin Development](https://wiki.librecad.org/index.php?title=LibreCAD_Plugin_Development).
For example, how do I write the plugin code to:
- Get text input from the user?
- Obtain the input coordinates?
Additionally, how do I write the code to:
- Get all objects and their related attributes (such as coordinates, length, enclosed area, enclosed circular area, filled area of the pattern, etc.) in the current user's selection set?
Furthermore, I would like to know how to:
- Create a small button tool that, when clicked, pops up a custom interface that I've defined.

So, where can I find more detailed documentation for plugin development?Or videos with rich plugin development examples?

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info
2.2.0

dxli, 2024-08-05:
One quick suggestion, please try the development version. 2.2.0 is in maintenance mode only, not in active development

c61292558, 2024-08-05:
> One quick suggestion, please try the development version. 2.2.0 is in maintenance mode only, not in active development

But why can't I find the development version on the homepage?

dxli, 2024-08-05:
The branches:

master ( 2.2.2 in the future ): the latest in active development 
2.2.1: the latest release candidate, under feature freeze 
2.2.0: stable release, not in active development

dxli, 2024-08-05:
I am not clear on plugin development, but reading through the existing first. I think you may also try qml for plugin.

We have plugins require selection, and GUI(plot equation).

c61292558, 2024-08-05:
qml？it‘s on QT6.  Is LibreCAD developing in QT5 ,right? 
> I am not clear on plugin development, but reading through the existing first. I think you may also try qml for plugin.
> 

I don't quite understand the meaning of this sentence：
> We have plugins require selection, and GUI(plot equation).

c61292558, 2024-08-05:
> The branches:
> 
> master ( 2.2.2 in the future ): the latest in active development 2.2.1: the latest release candidate, under feature freeze 2.2.0: stable release, not in active development

this link?https://github.com/LibreCAD/LibreCAD/releases
right?
 2.2.1_rc-latest  version ?

dxli, 2024-08-05:
I am talking about git branches.

The latest (2.2.2) is developed in qt6 only, but qml is available for qt6 as well. Even you are running qml, the actual plugin call to librecad must be by c++.

Read through our plugin source code, and you can find examples of selection and dialog creation.

I never tried to create a button inside of librecad from plugin. It's probably impossible

app4soft, 2024-08-06:
> I am talking about git branches.
> 
> The latest (2.2.2) is developed in qt6 only, but qml is available for qt6 as well. Even you are running qml, the actual plugin call to librecad must be by c++.

@c61292558, here is the link to the latest `2.2.2-alpha` builds with Qt6:
- https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.2_alpha-latest
Guess, `master` branch is exactly with Qt6, and all of Qt5 is left in `2.2.1` branch.
(correct me if I'm wrong)

dxli, 2024-08-06:
I am not aware of a plugin documentation for librecad.

Feel free to add one.

c61292558, 2024-08-07:
> I am not aware of a plugin documentation for librecad.
> 
> 我不知道一个插件留档librecad。
> 
> Feel free to add one.
Where to add?

lordofbikes, 2024-08-07:
> Where to add?

https://dokuwiki.librecad.org/

probably related: https://forum.librecad.org/I-need-detailed-documentation-on-developing-LibreCAD-plugins-tp5725201.html

closed 2024-08-07 by lordofbikes