LibreCAD/LibreCAD #1824: Exclude Zoom In/Out from affecting document change status

app4soft, 2024-08-12:
### Expected behavior

Zoom In/Out should not change document edit status.

Otherwise there should be an option in setting to exclude zoom actions from affecting document status.

### Observed behavior

Actually if document just saved (or just opened) and then zoom it document status changes - it is indicated in the title with asterisk (`*`).

### Steps to reproduce or sample file

1. Create and save document (or just open document);
2. Zoom it (without any edit action);
3. Look on window title.

### Operating System and LibreCAD version info

- **Version:** 2.2.2_alpha1-128- gf43e19c8
- **Compiler:** GNU GCC 9.4.0
- **Compiled on:** Aug 11 2024
- **Qt Version:** 6.6.3
- **Boost Version:** 1.71.0
- **System:** Debian GNU/Linux 12 (bookworm)

lordofbikes, 2024-08-13:
This is intended behavior. The viewport is part of the DXF file and thus zoom or panning modifies the file content.

This was already discussed here or in the forum, but couldn't find it yet with a quick search.

So the current behavior should stay the default and there could be an option in Application Preferences, to prohibit modifying by zoom, pan or maybe other actions too.

But imagine you want to share a detail of a drawing, so zoom into the detail, save the drawing and share it. Everyone who opens the drawing will see the detail immediately.
With the option activated, this works no longer, because when zoom/pan doesn't modify the drawing, it's not possible to save the current view without modifying the drawing somehow.

app4soft, 2024-08-13:
> So the current behavior should stay the default and there could be an option in Application Preferences, to prohibit modifying by zoom, pan or maybe other actions too.

Yes, please, add such option. Maybe, it should be in both "Application Preferences" and "Current Drawing Preferences":

- "Application Preferences" > "Defaults"
  - Save current viewport and zoom level in DXF: `yes` / `no`
- "Current Drawing Preferences" > "Paper":
  - Save current viewport and zoom level in DXF: `default`/`yes`/`no`

> imagine you want to share a detail of a drawing, so zoom into the detail, save the drawing and share it. Everyone who opens the drawing will see the detail immediately

This is good only for the first open of such DXF, because if someone would open it and occasionally re-saved it with changed zoom all this would lost for the 3rd person.

I would prefer create "frame" rectangle on separate later, so user navigate could navigate to such "frame" by "Fit all" (while hide other layers temporary) or by Fit to selected (if just select "frame" rectangle). Using such "frames" its possible store many viewports.

BTW, in DXF spec there is `viewport` entity and there could many viewports in a single DXF, but how it handled in LibreCAD?

- https://ezdxf.readthedocs.io/en/stable/dxfentities/viewport.html

Potentially, there might be an option for exported PDF to include viewports too:

- https://discuss.logseq.com/t/pdf-custom-viewport-mode-and-adaptive-dpi/2068
- https://stackoverflow.com/questions/47772334/how-do-i-extract-viewport-from-a-pdf-and-modify-an-annotations-bounding-rectang

PDF viewports export feature might be also related to https://github.com/LibreCAD/LibreCAD/issues/1823

sand1024, 2024-08-22:
Well, it seems that support of viewports is quite required feature and I'm thinking about adding it... 
However, in general it may be implemented by several ways. 
For example, there might be something like paper space and viewports are used mostly for organizing printing. 
Another possible option - viewports/named views are used for quick navigation between different areas of drawing (by storing corner position and zoom). 
What is the most convenient usage of viewports by your opinion? Could you please share some thoughts regarding them? 
Thank you.

app4soft, 2024-08-22:
> Another possible option - viewports/named views are used for quick navigation between different areas of drawing (by storing corner position and zoom).

This.

Also, this is a part of feature request on adding viewports export to PDF with interactive Index of viewports, see:

- https://github.com/LibreCAD/LibreCAD/issues/1823

u2fly, 2025-04-21:
Solved with option __"Application Preferences > Defaults > Changing view only changes document"__

closed 2025-04-21 by u2fly