LibreCAD/LibreCAD #1826: Crash after click on folder in Library Browser

app4soft, 2024-08-12:
### Expected behavior

Should not crash when open folders in Library Browser.

### Observed behavior

Crashed after click on any folder listed in Library Browser (on initializing thumbs/preview generation).

### Steps to reproduce or sample file
1. Launch LibreCAD;
2. Go to Library Browser panel and click on folder in the list;
3. Crash.

<details><summary>Crash Log:</summary>

```Shell
$ ./LibreCAD-2.2.2_alpha1-128-gf43e19c8.AppImage
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
/tmp/.mount_LibreCJBNODp/usr/share/librecad/qm
/tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts
/tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/amiri-regular.lff:
base: amiri-regular
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/azomix.lff:
base: azomix
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/azomix_i.lff:
base: azomix_i
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/cursive.lff:
base: cursive
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/cyrillic_ii.lff:
base: cyrillic_ii
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/gothgbt.lff:
base: gothgbt
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/gothgrt.lff:
base: gothgrt
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/gothitt.lff:
base: gothitt
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/greek_ol.lff:
base: greek_ol
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/greekc.lff:
base: greekc
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/greekcs.lff:
base: greekcs
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/greekp.lff:
base: greekp
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/greeks.lff:
base: greeks
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/iso.lff:
base: iso
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/iso3098.lff:
base: iso3098
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/iso3098_i.lff:
base: iso3098_i
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/italicc.lff:
base: italicc
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/italiccs.lff:
base: italiccs
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/italict.lff:
base: italict
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/kochigothic.lff:
base: kochigothic
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/kochimincho.lff:
base: kochimincho
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/kst32b.lff:
base: kst32b
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/lc_opengost-ar.lff:
base: lc_opengost-ar
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/lc_opengost-br.lff:
base: lc_opengost-br
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/opengosttypea-regular.lff:
base: opengosttypea-regular
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/opengosttypeb-regular.lff:
base: opengosttypeb-regular
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/romanc.lff:
base: romanc
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/romancs.lff:
base: romancs
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/romand.lff:
base: romand
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/romanp.lff:
base: romanp
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/romans.lff:
base: romans
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/romansi.lff:
base: romansi
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/romant.lff:
base: romant
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/scriptc.lff:
base: scriptc
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/scripts.lff:
base: scripts
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/simplex.lff:
base: simplex
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/standard.lff:
base: standard
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/syastro.lff:
base: syastro
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/symap.lff:
base: symap
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/symath.lff:
base: symath
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/symbol.lff:
base: symbol
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/symbol_misc1.lff:
base: symbol_misc1
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/symbol_misc2.lff:
base: symbol_misc2
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/symeteo.lff:
base: symeteo
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/symusic.lff:
base: symusic
font: /tmp/.mount_LibreCJBNODp/usr/share/librecad/fonts/unicode.lff:
base: unicode
/tmp/.mount_LibreCJBNODp/usr/share/librecad/patterns
/tmp/.mount_LibreCJBNODp/usr/share/librecad/qm
QFSFileEngine::open: No file name specified
qt.core.qobject.connect: QObject::connect: No such signal LC_QuickInfoWidget::escape()
qt.core.qobject.connect: QObject::connect:  (sender name:   'LC_QuickInfoWidget')
QLayout: Attempting to add QLayout "" to QG_LibraryWidget "Library", which already has a layout
/tmp/.mount_LibreCJBNODp/usr/share/librecad/library
/tmp/.mount_LibreCJBNODp/usr/lib/librecad
0
qt.core.qobject.connect: QObject::connect: No such signal QC_MDIWindow::signalClosing(QC_MDIWindow*)
/tmp/.mount_LibreCJBNODp/usr/share/librecad/library
/tmp/.mount_LibreCJBNODp/usr/share/librecad/library
/tmp/.mount_LibreCJBNODp/usr/share/librecad/library
/tmp/.mount_LibreCJBNODp/usr/share/librecad/library
0
0
Segmentation fault
```

</details>

### Operating System and LibreCAD version info

- **Version:** 2.2.2_alpha1-128-gf43e19c8
- **Compiler:** GNU GCC 9.4.0
- **Compiled on:** Aug 11 2024
- **Qt Version:** 6.6.3
- **Boost Version:** 1.71.0
- **System:** Debian GNU/Linux 12 (bookworm)

### Comment

As a part of solution, it would be good if LibreCAD would has an option to disable thumbs/preview (and generation) in Library Browser, and if tumbs toggled off show folders content as list of files (without previews).

dxli, 2024-08-17:
I cannot reproduce this issue.

Could you find steps to reproduce issue?

app4soft, 2024-08-17:
> Could you find steps to reproduce issue?

As described in first comment just do:

1. Launch latest `2.2.2_alpha` LibreCAD AppImage;
2. Move mouse cursor on Library Browser widget;
3. Click on any folder in Library Browser;
4. Crash.

<details><summary>Crash Log:</summary>

```gdb
$ LibreCAD-2.2.2_alpha1-134-g8a07be76 -d 6
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
reading  6  as debugging level
RS_DEBUG::setLevel(6)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
RS_DEBUG: Notice
RS_DEBUG: Informational
RS_DEBUG: Debugging
param 0: ./LibreCAD-2.2.2_alpha1-134-g8a07be76.AppImage
appDir2:/tmp/.mount_LibreCKnEdGL/usr/bin

RS_System::init: System LibreCAD initialized.
RS_System::init: App dir: /tmp/.mount_LibreCKnEdGL/usr/bin
RS_System::initLanguageList
RS_System::getFileList: subdirectory qm 
RS_System::getFileList: appDirName librecad 
RS_System::getFileList: getCurrentDir /home/user/Downloads 
getDirectoryList(): line 602: dir=/tmp/.mount_LibreCKnEdGL/usr/bin

getDirectoryList(): line 623: dir=/home/user/.librecad/qm

getDirectoryList(): line 623: dir=/tmp/.mount_LibreCKnEdGL/usr/share/doc/librecad/qm

getDirectoryList(): line 623: dir=/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm

getDirectoryList(): line 623: dir=/tmp/.mount_LibreCKnEdGL/usr/lib64/librecad/qm

getDirectoryList(): line 623: dir=/tmp/.mount_LibreCKnEdGL/usr/lib/librecad/qm

RS_System::getDirectoryList: Paths:
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm
getDirectoryList(): line 685: dir=/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm

getFileList():: fileList:
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_ar.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_ca.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_cs.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_da.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_de.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_el.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_en.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_en_au.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_es.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_es_ar.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_es_bo.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_es_cl.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_es_co.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_es_cr.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_es_do.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_es_ec.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_es_gt.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_es_hn.qm
/tmp/.mount_LibreCKnEdGL/usr/share/librecad/qm/librecad_es_mx.qm
[... 3614 lines cut ...]
RS_EntityContainer::updateInserts: skip entity ID/type: 7/8
RS_EntityContainer::updateInserts: skip entity ID/type: 7/8
RS_EntityContainer::updateInserts: skip entity ID/type: 7/8
RS_EntityContainer::updateInserts: skip entity ID/type: 7/8
RS_EntityContainer::updateInserts: skip entity ID/type: 7/8
RS_EntityContainer::updateInserts() ID/type: 7/8
RS_EntityContainer::updateInserts() ID/type: 5/5
RS_FilterDXFRW::fileImport OK
RS_FilterDXFRW::~RS_FilterDXFRW(): OK
RS_Graphic::open(/tmp/.mount_LibreCKnEdGL/usr/share/librecad/library/block/block10.dxf): OK
RS_GraphicView::zoomAuto
RS_EntityContainer::calculateBorders
RS_EntityContainer::calculateBorders
RS_EntityContainer::calculateBorders: size 1: 36.373066,40.000000
RS_EntityContainer::calculateBorders: size: 36.373066,40.000000
RS_EntityContainer::calculateBorders: size 1: 36.373066,40.000000
RS_EntityContainer::calculateBorders: size: 36.373066,40.000000
f: 3.244159/2.950000
f: 2.950000/2.950000
f: 2.950000/2.950000
RS_GraphicView::zoomAuto OK
Segmentation fault
```

<details>

Questions2000, 2024-08-31:
Ahh, this is another one that you have mentioned that I came across as well.

Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8
LibreCAD-2.2.2_alpha1-121-g512680e6.AppImage

Same issue, the second you double click on a folder LC closes instantly.

closed 2024-08-31 by app4soft