LibreCAD/LibreCAD #1833: Offset & Parallel conflict with one another

jhmonroe, 2024-08-19:
### Expected behavior

Offset should allow offset creation for any shape, not just straight lines.

### Observed behavior

Observe the commands that appear in command line when you select the following options:

- Modify > Offset = Offset: offset, o (this command will allow you to offset a rectangle only when selected using the button)
- Lines > Parallel = Command: o (offset) (this command will only allow you to add parallel lines to straight lines when using the button)
- The key commands pa or o are overlapping/shared and also only activate the Parallel version of the offset command
- Thus it is not possible to activate the true Offset tool using key commands

### Steps to reproduce or sample file

### Operating System and LibreCAD version info
Mac OS
LC 22.0.2

dxli, 2024-08-19:
Offset is a feature needs development, just like dimensions and hatches

dxli, 2024-08-21:
Pushed a fix to master:

line parallel: linepar parallel lineoff; pa ll
offset: modoffset offset; mo moff

dxli, 2026-08-25:
PR #2513 has been reworked and now resolves this directly.

`offset` is kept on the real Modify > Offset action (`ActionModifyOffset`), not moved back to the parallel-line tool - moving it there would reintroduce exactly this bug, since `RS_Creation::createParallel` only handles lines/arcs/circles/parabolas and has no polyline support, while `RS_Polyline::offset()` is implemented and reachable through the real Offset action. AutoCAD, BricsCAD and DraftSight all define OFFSET over polylines too.

Also restored the `o` short command on Modify > Offset, which has been unbound since the original fix for this issue (bedbe790d) dropped it from the parallel tool without giving it a new home - so the toolbar button's own "Offset : offset, o" hint has been wrong for about two years.

Leaving this open until #2513 actually merges.

dxli, 2026-08-25:
Fixed by PR #2513 (merge a05b4261f): offset stays bound to the real Modify > Offset action, and the o short command is restored, so the Offset toolbar button's own command hint ("Offset : offset, o") now actually works.

closed 2026-08-25 by dxli