LibreCAD/LibreCAD #1835: LibreCAD ignores font defined U+FFFD symbol shape

app4soft, 2024-08-21:
### Expected behavior

LibreCAD should use shapes of all symbols set defined in LFF.

If font has custom shape for `[fffd] �` (U+FFFD) symbol LibreCAD should use it.

### Observed behavior

Actually LibreCAD uses hardcoded shape for `[fffd] �` for all fonts, ignoring font defined shape.

Discovered after https://github.com/LibreCAD/LibreCAD/pull/1834 has been merged.

### Steps to reproduce or sample file

Here is a testcase LFF-font sample (copy code and save as LFF file, then open in LibreCAD).

There is `[fffd] �` symbol copied shape from `[003f] ?` symbol, and then itself copied to `[00bf] ¿` symbol.

```
# Format:            LibreCAD Font 1
# Creator:           LibreCAD
# Version:           2.2.2_alpha1-138-g515194f8
# Name:              LFF
# Encoding:          UTF-8
# LetterSpacing:     3
# WordSpacing:       6.75
# LineSpacingFactor: 1
# Created:           
# Last modified:     2024-08-21
# License:           Public Domain

[fffd] �
C003f

[003f] ?
3,0;0,3
0,3;0,6
0,6;3,9
3,9;6,6
6,6;6,3
6,3;3,0
4.2,5.4;1.8,5.4,A1
4.2,5.4;3.96822,4.75982,A-0.17545
3.3,4.1;2.99957,3.40001,A0.19789
3.97,4.76;3.3,4.1
3,2.8;3,2.2,A1
3,2.2;3,2.8,A1

[00bf] ¿
Cfffd
```

And here is how it looks in LibreCAD

[image]

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-138- g515194f8
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 21 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

dxli, 2024-08-21:
fffd is the "replacement character"

It's intended to be an unknown character for the font. Should a font define its own unknown? Or the software should assign the unknown character if the font doesn't know a character?

Let a font know about the unknown is the issue. How do I know whether the shape is unknown to the font?

app4soft, 2024-08-21:
> Let a font know about the unknown is the issue.

Yes and no.

In general fonts design (TTF, OTF) font designers defines the shape of unknown (U+FFFD) symbol.

So, if font has this symbol shape defined, and has been assigned to some text object then LibreCAD should use shape from this font for unknown characters in such text object.

If used font has no shaep defined for U+FFFD, then LibreCAD should use hardcoded `[fffd] �` rhombus shape.

Also, I recommend to use add simplified question mark inside hardcoded rhombus.

New shape of `[fffd] �` actually merged into `unicode.lff` is a bit complex for use as hardcoded, as it has 5 arcs and 7 line segment (instead of hardcoded 4 line segment rhombus; question mark for hardcoded could be added with just 4 extra line segments).

lordofbikes, 2024-08-22:
This would just make support harder.
Undefined character is an often asked question in the forum and our valued supporting members can answer these questions easy and confident.

But when LibreCAD would show font defined symbols, the supporters have to check the font first, to be sure the issue is caused by missing characters in the font.
Possibly the supporters don't even have access to the font in question, what makes it impossible for them to help.

So it's a nice feature, but the goal of this symbol is, to never see it.
If the font creator would support the users I'd agree. But when LibreCAD users detect an issue with the rhombus symbol, they contact us and thus it is much more comfortable and time saving if we keep to ignore font defined `[fffd]` symbols.

ruevs, 2024-08-22:
(I'm not familiar with LibreCAD but) I agree with the @lordofbikes's reasoning.

I created https://github.com/LibreCAD/LibreCAD/pull/1834 because [while I was updating unicode.lff in SolveSpace](https://github.com/solvespace/solvespace/pull/1478) I decided to take the latest changes from LibreCAD and I noticed we had two changes, that were not upstream-ed.

SolveSpace only has one vector font (unicode.lff) used to render all text in the OpenGL view that is not part of the model. [The `�` character from the font is used to display unknown characters. This ](https://github.com/solvespace/solvespace/commit/cd6d8911009362a39575f4ad368a3cad68386e5c) is why the nicer glyph was added.

So in SolveSpace there is no ambiguity what is happening when `�` appears but for LibreCAD @lordofbikes's reasoning makes sense to me - a single "unique" symbol should be displayed for a missing character from any font.

closed 2024-08-22 by app4soft