LibreCAD/LibreCAD #1838: Can not display Chinese words

qdlzc, 2024-08-22:
### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info
[image]

![Can not Display Chinese words]

(https://github.com/user-attachments/assets/241edc3d-be86-41ea-a305-31492449a101)

dxli, 2024-08-29:
Right now, Chinese fonts are not included in the official packages.

You can download the WenQuanYi fonts for Chinese:

https://dokuwiki.librecad.org/doku.php/community:fonts

Alternatively, you may try our ttf2lff tool to create your own Chinese lff fonts from TTF fonts

app4soft, 2024-08-29:
As WenQuanYi is outlined font created using ttf2lff, there is alternative single line (stroke, stencil) Hershey font:

- https://github.com/LingDong-/chinese-hershey-font

To convert [`Mingti.hf.txt`](https://github.com/LingDong-/chinese-hershey-font/blob/master/dist/hershey/Mingti.hf.txt) font file to LFF, there is a Python script: [`jhftool`](https://github.com/user-attachments/files/16647025/jhftool.txt) (rename to `jhftool.py`)