LibreCAD/LibreCAD #1842: LibreCAD crashes on RMB click in toolbars area

app4soft, 2024-08-22:
### Expected behavior

There should not be crashes of LibreCAD.

### Observed behavior

After https://github.com/LibreCAD/LibreCAD/pull/1840 LibreCAD `2.2.2_alpha` starting crashing if click RMB in toolbars area.

In previos LibreCAD `2.2.2_alpha` build there was no such issue.

### Steps to reproduce or sample file

1. Launch LibreCAD;
2. Move cursor on any toolbar;
3. Click RMB;

Its results in crash.

Debug log (`-d 6` level):

<details><summary>Debug Log:</summary>

```Bash
$ ./LibreCAD-2.2.2_alpha1-140-g05c0f3bd.AppImage -d 6
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
reading  6  as debugging level
RS_DEBUG::setLevel(6)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
RS_DEBUG: Notice
RS_DEBUG: Informational
RS_DEBUG: Debugging
param 0: ./LibreCAD-2.2.2_alpha1-140-g05c0f3bd.AppImage
appDir2:/tmp/.mount_LibreCNHckKF/usr/bin

RS_System::init: System LibreCAD initialized.
RS_System::init: App dir: /tmp/.mount_LibreCNHckKF/usr/bin
RS_System::initLanguageList
RS_System::getFileList: subdirectory qm 
RS_System::getFileList: appDirName librecad 
RS_System::getFileList: getCurrentDir /home/user/AppImages 
getDirectoryList(): line 602: dir=/tmp/.mount_LibreCNHckKF/usr/bin

getDirectoryList(): line 623: dir=/home/user/.librecad/qm

getDirectoryList(): line 623: dir=/tmp/.mount_LibreCNHckKF/usr/share/doc/librecad/qm

getDirectoryList(): line 623: dir=/tmp/.mount_LibreCNHckKF/usr/share/librecad/qm

getDirectoryList(): line 623: dir=/tmp/.mount_LibreCNHckKF/usr/lib64/librecad/qm

getDirectoryList(): line 623: dir=/tmp/.mount_LibreCNHckKF/usr/lib/librecad/qm

RS_System::getDirectoryList: Paths:
/tmp/.mount_LibreCNHckKF/usr/share/librecad/qm
getDirectoryList(): line 680: dir=/tmp/.mount_LibreCNHckKF/usr/share/librecad/qm

getFileList():: fileList:
/tmp/.mount_LibreCNHckKF/usr/share/librecad/qm/librecad_ar.qm
/tmp/.mount_LibreCNHckKF/usr/share/librecad/qm/librecad_ca.qm
/tmp/.mount_LibreCNHckKF/usr/share/librecad/qm/librecad_cs.qm
/tmp/.mount_LibreCNHckKF/usr/share/librecad/qm/librecad_da.qm
/tmp/.mount_LibreCNHckKF/usr/share/librecad/qm/librecad_de.qm
/tmp/.mount_LibreCNHckKF/usr/share/librecad/qm/librecad_el.qm
/tmp/.mount_LibreCNHckKF/usr/share/librecad/qm/librecad_en.qm
/tmp/.mount_LibreCNHckKF/usr/share/librecad/qm/librecad_en_au.qm
[... 1055 lines cut ...]
RS_Snapper::catchEntity
RS_EntityContainer::getNearestEntity
RS_EntityContainer::getDistanceToPoint
RS_EntityContainer::getDistanceToPoint: OK
RS_EntityContainer::getNearestEntity: OK
RS_Snapper::catchEntity: not found
RS_Snapper::catchEntity
RS_EntityContainer::getNearestEntity
RS_EntityContainer::getDistanceToPoint
RS_EntityContainer::getDistanceToPoint: OK
RS_EntityContainer::getNearestEntity: OK
RS_Snapper::catchEntity: not found
Segmentation fault
```

</details>

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-140-g05c0f3bd
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 22 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-08-22:
hm... a bit weird yet interesting... sure, I'll take a look - thank you for reporting.

sand1024, 2024-08-22:
I've committed quick fix, please check as it will be merged into master

app4soft, 2024-08-22:
> please check as it will be merged into master

I can check only when it would be merged into master, as I use continuous builds binary, as there is no artifacts produced for open pull requests.

app4soft, 2024-08-22:
Tested. Fixed.

closed 2024-08-22 by app4soft