LibreCAD/LibreCAD #1844: Add artifact storing for Actions jobs for open pulls

app4soft, 2024-08-22:
### Expected behavior

Actions should produce and store artifacts for pulls (at least just AppImage), without upload to releases assets.

### Observed behavior

Actually Actions workflow set to build from source, but drops and do not store artifacts for pulls.

### Steps to reproduce or sample file

1. Check any open pull Actions job artifacts section.

### Operating System and LibreCAD version info

GitHub Workflow issue/feature request.

lordofbikes, 2024-08-23:
It is not clear to me what this is about.

I suspect you expect, that pull requests create binaries, which can be downloaded?
If so, this is not recommend for various reasons.

Also, the actions run on pull request are not to build binaries. They trigger the `CodeQL` workflow which runs some general checks across the code to detect obvious issues.

app4soft, 2024-08-23:
> I suspect you expect, that pull requests create binaries, which can be downloaded?
> Also, the actions run on pull request are not to build binaries.

Its possible set workflow to build artifact for action run on open pull request:

- https://github.com/marketplace/actions/pull-request-artifact

This should create binary and store it as job artifact on jobe page in Actions area (GitHub stores it for a month or so, and then autoremoves it).

Here is an example of open pull request whre action runs built binary artifacts (artifacts at the bottom, and already expired):
- https://github.com/YSCEDC/YSCE/actions/runs/6443037817/

Take a look on workflow setup
- https://github.com/YSCEDC/YSCE/blob/master/.github/workflows/continuous.yml

> If so, this is not recommend for various reasons.

Actions artifacts are available only for GitHub logged-in users, so regular LibreCAD user would not be able to download such experimental builds.

closed 2025-04-14 by u2fly