LibreCAD/LibreCAD #1845: Mtext kerning: LibreCAD-2.2.1_rc3-16-g0a0a81de-win64.exe

KoTCiB, 2024-08-23:
### Expected behavior
Mtext and Text should produce similar results with a single line of text when using outline fonts.

### Observed behavior
MText has wrong kerning  regardless of whether one uses Right to Left or Left to Right

### Steps to reproduce or sample file
I created an outline font from ariel.ttf that comes with my Windows 11.  I am enclosing the test.dxf file and also a screen capture of the result.  I did not include ariel.lff

### Operating System and LibreCAD version info
Windows 11, LibreCAD-2.2.1_rc3-16-g0a0a81de-win64.exe 
[MText-LibreCAD.zip](https://github.com/user-attachments/files/16724398/MText-LibreCAD.zip)

app4soft, 2024-08-23:
Issue seems specific to spaces between words and some punctuation marks symbols kerning (see full-width comma overlapping, it is latest LC `2.2.2_alpha`), but there is no issues for letters' kerning (words without spaces and punctuation marks looks the same in both Text and MText).

Here are two pairs of strings (upper string in each pair is Text, and lower string is MText).

1. First pair uses `�，М 1:10 ЇїҐґЄє` (there is full-width comma, implemented in https://github.com/LibreCAD/LibreCAD/pull/1834)
2. Second pair uses `@ AbCdEfGe 12345689.0`

Both Text and MText uses `unicode.lff` font and has been set letter size to `5`, and line spacing to `1`.

[image]

Version: 2.2.2_alpha1-141-g0e0b66e8
Compiler: GNU GCC 9.4.0

- Compiled on: Aug 22 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

app4soft, 2024-08-23:
Probably related (or duplicate) of:
- https://github.com/LibreCAD/LibreCAD/issues/1818

KoTCiB, 2024-08-23:
Please note that at least in the example reported earlier, it is kerning between characters inside words.  Specifically, additional space is displayed prior to the last 2 characters in the Hebrew language.