LibreCAD/LibreCAD #1846: Performance regression in text rendering

app4soft, 2024-08-23:
### Expected behavior

There should not be performance regressions after new features implemented.

### Observed behavior

Actually there is a huge performance regression in text rendering, affected by merged https://github.com/LibreCAD/LibreCAD/pull/1840 and https://github.com/LibreCAD/LibreCAD/pull/1843.

### Steps to reproduce or sample file

1. Load latest LibreCAD `2.2.2_alpha`;
2. Open document (or create new one) with few Text and MText labels and some dimensions;
3. Try pan and zoom.

Sample DXF: [A4_Landscape_ISO7200TD_dxf.zip](https://github.com/user-attachments/files/16576202/A4_Landscape_ISO7200TD_dxf.zip)

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-141-g0e0b66e8
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 22 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-08-23:
That's really strange.. I can't see the regression you've mentioned on the file you've provided. Both zoom and pan work fine - yet, I'm on Windows, and probably such behavior is OS-specific. 
What is regression? Drawing speed? Zoom step? etc. 
I'll check this, yet I think we'll review rendering flow soon.

app4soft, 2024-08-24:
> What is regression? Drawing speed? Zoom step?

Drawing speed on Zoom in/out and Scroll up-down/left-right (as on pan/drag LibreCAD simplifies text rendering to just rectangle until pan/drag stops).

> That's really strange.. I can't see the regression you've mentioned on the file you've provided. Both zoom and pan work fine - yet, I'm on Windows, and probably such behavior is OS-specific.

Probably, because your PC is modern and has more powerfull CPU and more RAM.

Try to create drawing with a huge text sample (like _"Lorem ipsum dolore..."_ with few hundreds or thousands of characters) and try to zoom in/out and scroll left-right/up-down with LibreCAD builds before and after new featers merged.

I suppose development builds of LibreCAD should be always tested on low-end PC or on virtual machine with minimal settings (CPU <= 2GHz; RAM <= 2Gb), where its easy to see and detect regressions.

app4soft, 2024-08-25:
> That's really strange.. I can't see the regression you've mentioned on the file you've provided.

I might lockated a bit of specific: performance dropped object has applied line style with defined width.

For any object with `Default` and `0.00mm` line styles there is no performance drop.

Performance starts drops when LirbeCAD starts draw objects' line styles "in width" (i.e. Zoom In close enough to see difference between `Default`/`0.00mm` and other line width styles in drawing).

There was probably no such issue before https://github.com/LibreCAD/LibreCAD/pull/1840 has been merged (I suppose it bringed some extra rendering job for objects lines when it is non-zero width on screen).

If Zoom Out enough to LibreCAD starts draw all lines in minimal width there is performance increased drastically.

app4soft, 2024-08-27:
Additional details: tried the same file ([A4_Landscape_ISO7200TD_dxf.zip](https://github.com/user-attachments/files/16576202/A4_Landscape_ISO7200TD_dxf.zip)) in `LibreCAD-2.2.1_rc3-16-g0a0a81de-x86_64.AppImage` — and there is no such performance issue.

Even [lorem-ipsum_dxf.zip](https://github.com/user-attachments/files/16741593/lorem-ipsum_dxf.zip) in latest `LibreCAD-2.2.1_rc3` has no such a huge laggs, comparing to latest `LibreCAD-2.2.2_alpha`.

@sand1024, I suppose it needs to review https://github.com/LibreCAD/LibreCAD/pull/1840 for increased use of resources on rendering non-zero width lines.

sand1024, 2024-08-28:
well, for sure I'll check this - the only thing that I didn't directly modify rendering pipeline yet, so it seems that such behavior is some side effect... 
I have a couple of ideas what it may be, yet need some time to test and check.

dxli, 2024-08-28:
Cap and connection style could be expensive.

app4soft, 2024-08-29:
With https://github.com/LibreCAD/LibreCAD/pull/1865 issue almost dissapear, except of the case when objects higlihghted with mouse cursor during zoom in/out.

Please, review highlighting (red color preview?) code too.

closed 2025-04-18 by u2fly