LibreCAD/LibreCAD #1847: Red solid color fill in full screen in drawing area on hovering Min/Max/Center on Entitiy info

app4soft, 2024-08-23:
### Expected behavior

There should not be red screens.

### Observed behavior

If hover `Min`/`Max`/`Center` on "Entity info" dockwidget there is full screen solid #red color fill in drawing area.

With "Draft mode" turned ON there is no such issue.

### Steps to reproduce or sample file

1. Launch latest LibreCAD `2.2.2_alpha`
2. Create drawing with any entities;
3. Select one entity and move cursor to "Entity info" dockwidget;
4. Hover cursor over  `Min`/`Max`/`Center` values on "Entity info";
5. Try turn ON and OFF "Draft mode" and repeat previous step.

Screenshot:

[image]

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-141-g0e0b66e8
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 22 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-08-23:
couple of questions: 
1) which color is used for highlighting points? (please check by clicking settings icon in settings for Entity Info widget for color). Is it red or something different? 
2) Is this reproduceable all times or right after opening the file? More interesting, could you, say, open the same file, draw a single point or line of points and try hovering over coordinate again? Will you experience the same behavior?

app4soft, 2024-08-23:
1. Here is screenshot with color setinngs:

[image]

2. Fully reproducible, with any file, even just with new file after startup. Note that it dissapears in Draft mode, but its back if disable Draft mode.

Here i screencast:

[simplescreenrecorder-2024-08-24_02.37.34.webm](https://github.com/user-attachments/assets/e68b2547-42cf-43d7-8818-ba3aadae47b5)

sand1024, 2024-08-24:
oh, thank you, will check. 
Could you please also check which color is used for highlight points using Entities Info settings dialog - it's invoked by grid gear arrow at the top right corner of Entities Info widget toolbar.  
Is it red?

app4soft, 2024-08-24:
> Could you please also check which color is used for highlight points
> Could you please just add screenshot for that setting dialog?

Its "Red" (`#ff0000`), of course. (I`ve never touched this setting, so suppose it is default configuration)

[image]

> Also, it will be really helpful, if you could change settings for Pen used for highlighting points and check whether using non-default settings affect such behavior.

After changing color to another (and even then back it to Red) **issues dissapears for actual LibreCAD session only**.

After relaunch LibreCAD issue is back. Suppose:
1. Its might be some missing configuraion, not storing (and LibreCAD resets to default full screen solid fill).
2. Such full screen solid fill should not exists in the first place.

sand1024, 2024-08-24:
thank you for providing this info - it was really helpful. Finally was able to reproduce and fix that bug. 

Indeed, the issue was related to settings, and more specifically, to storing "Default" value of linetype there. 

Surprisingly, there was an old bug quite deeply within settings subsystem, that incorrectly read that value - and so the point was drawn using MAX_INT line width... it's weird enough. 

Anyway, I've found the reason and fixed it, so as soon as I'll commit and it will be merged, this issue will gone. Hope it will be one of these days. 

For now, you can just change the linetype for points in settings dialog to any value except "default". 

BTW the shape of highlighting point is controlled by settings for Drawing.

Questions2000, 2024-08-31:
Ahh I was going to report this exact same issue, thank you for reporting and also thank you for taking care of the issue.

closed 2025-04-14 by u2fly