LibreCAD/LibreCAD #1848: Key tip extends width of Cmd dockwidget in Print preview mode

app4soft, 2024-08-23:
### Expected behavior

Dockwidgets width and height should not be affected by any content shown in a such dockwidget.

### Observed behavior

"Cmd" dockwidget affected by default key tip in "Print preview" mode, which extends its width.

### Steps to reproduce or sample file
1. Launch LibreCAD;
2. Minimize "Cmd" dockwidget width;
3. Switch to "Print preview" mode and look on "Cmd" width.

Screenshot:

[image]

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-141-g0e0b66e8
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 22 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-08-23:
well, sizing policy for widgets in the left sidebar is.. well, a bit complicated, especially for small screen resolutions as we have to fit into quite a limited space. 

Expanding command widget to fit the message is actually default behavior of cmd widget. 

Basically, there we should use some kind of trade-off - either use short and less informative messages or expand widgets. 
For example, it's possible either truncate text or add some ellipsis at the end - but that will also truncate the meaning of the message. 

I can't say that this is really "an issue" - but probably that might be improved via some hint or something.. I'll think about this.

app4soft, 2024-08-24:
> Basically, there we should use some kind of trade-off

Does action/shortcut message should be drawn in Cmd title at all?

Also, what about draw such message line just in working area at the bottom center? (similar to how "Draft" labels are drawn in all corners while Draft mode is turned on)

u2fly, 2025-04-18:
> make it optional to show/hide.

@sand1024 maybe add a checkbox option in Cmd dock options menu?

> Also, what about draw such message line just in working area at the bottom center? (similar to how "Draft" labels are drawn in all corners while Draft mode is turned on)

Already implemented in "Info Cursor > Promt"

sand1024, 2025-04-18:
@u2fly 
Yes, I'll take a look. Actually, the entire cmd widget needs a face lifting too.. 

Btw, according to you screen - you can simply switch status bar to floating mode and gain some space in the status bar area by re-arranging widgets there.

closed 2025-04-21 by u2fly