LibreCAD/LibreCAD #1850: Crash on "Cancel" click in resized wndow of Text/MText dialog

app4soft, 2024-08-23:
### Expected behavior

LibreCAD should not crash.

### Observed behavior

### Steps to reproduce or sample file
1. Launch LibreCAD;
2. Create new document;
3. Activate Text or MText tool;
4. Resize dialog window;
5. Click <kbd>Cancel</kbd>.

In a result LibreCAD crashes.

Debug Logs:

- Text crash: [text-resize-cancel.log](https://github.com/user-attachments/files/16726556/text-resize-cancel.log)
- MText crash: [mtext-resize-cancel.log](https://github.com/user-attachments/files/16726558/mtext-resize-cancel.log)

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-141-g0e0b66e8
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 22 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-08-23:
Well, could you please try to collect more information about the context? Say, are there other open documents, which widgets are on the screen etc..? 
I've checked this behavior in different variations - yet in my setup I can't reproduce it, so any additional insights will be helpful.

app4soft, 2024-08-23:
> Well, could you please try to collect more information about the context?

No, nothing is open.

Just start LibreCAD, activate Text or MText, resize its dialog window, press <kbd>Cancel</kbd> -- and its crashes.

Thats all.

Here is my `~/.config/LibreCAD/LibreCAD.conf`:
<details><summary>LibreCAD.conf</summary>

```
[General]
HatchPattern=
HatchPreview=1
HatchSolid=1

[Appearance]
Antialiasing=1
DraftMode=0
ExtendAxisLines=0
GridSpacingX=0
GridSpacingY=0
GridType=0
IsometricGrid=0
Language=en
LanguageCmd=en
RefPointSize=2
VisualizeHovering=0
indicator_lines_type=Crosshair

[ColorBox]
CustomColor0=2147483647
CustomColor1=2147483647
CustomColor2=2147483647
CustomColor3=2133864496

[Colors]
background=#000000

[CustomToolbars]
DefaultCustom=EditKillAllActions
ModifyProperty=ModifyAttributes, ModifyEntity, FilePrintPreview

[Defaults]
AutoBackupDocument=1
Unit=Millimeter
UseQtFileOpenDialog=1

[Draw]
ArcTangentialRadius=0.08
CircleByArcRadiusShift=0
CircleRadius=.5
DimDiameter=0
DimLabel=
DimTol1=
DimTol2=
LineAngleLength=10.00000000
LineAngleRelAngle=1
LineAngleRelAngleIsRelative=0
LineAngleRelLength=1
LineAngleRelOffset=1
LineAngleRelSnapDistance=0
LineBisectorLength=100
LineParallelDistance=1
LinePointsAngle=0
LinePointsPointsDistance=1
PolylineMode=1
SplineClosed=0
SplineDegree=1
[... 90 lines cut ...]
[Startup]
FirstLoad=0
ShowSplash=1
TabMode=1

[Widgets]
AllowStatusbarFontSize=0
AllowStatusbarHeight=1
AllowStyle=1
AllowTheme=0
AllowToolbarIconSize=1
KeycodeMode=false
LeftToolbarColumnsCount=5
StatusbarHeight=32
Style=Fusion
StyleSheet=
ToolbarIconSize=22

[WindowOptions]
SubWindowMode=0
TabPosition=1
TabShape=0
```

<details>

app4soft, 2024-08-23:
Here is screencast

[simplescreenrecorder-2024-08-24_02.31.51.webm](https://github.com/user-attachments/assets/f012093e-c54b-41a2-84c3-162ed5d9b87a)

app4soft, 2024-08-27:
Also, checked: crash happen on other dialogs too, including "Properties", "Attributes", "Hatch", "Drawing Preferences".

Even just move such floating dialog window (without resize) and click on <kbd>Cancel</kbd> results in LibreCAD crash.

But crash not happen on same actions for "Keyboard Shortcuts" and "Application Preferences" windows.

app4soft, 2024-08-31:
> Even just move such floating dialog window (without resize) and click on <kbd>Cancel</kbd> results in LibreCAD crash.

Crash also happens on <kbd>OK</kbd> and closing tab in window title.

dxli, 2024-09-01:
This is probably due to preview and resuming of the modify action.

A bisection can find first commit triggering this issue.

app4soft, 2024-10-21:
Here is another case of crash in the latest LibreCAD AppImage

### Steps to reproduce or sample file

1. Launch LibreCAD;
2. Create document;
3. Draw "Spline through points";
4. Double-click (or use "Properties" tool) on spline to open "SplinePoints";
5. Resize "SplinePoints" window and close by any, <kbd>Cancel</kbd> or <kbd>OK</kbd> buttons.
6. See error.

<details><summary>Crash Log</summary>

```Shell
$ ~/AppImages/LibreCAD
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
/tmp/.mount_LibreCKgmbaN/usr/share/librecad/qm
/tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts
/tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/amiri-regular.lff:
base: amiri-regular
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/azomix.lff:
base: azomix
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/azomix_i.lff:
base: azomix_i
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/cursive.lff:
base: cursive
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/cyrillic_ii.lff:
base: cyrillic_ii
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/gothgbt.lff:
base: gothgbt
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/gothgrt.lff:
base: gothgrt
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/gothitt.lff:
base: gothitt
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/greek_ol.lff:
base: greek_ol
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/greekc.lff:
base: greekc
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/greekcs.lff:
base: greekcs
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/greekp.lff:
base: greekp
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/greeks.lff:
base: greeks
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/iso.lff:
base: iso
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/iso3098.lff:
base: iso3098
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/iso3098_i.lff:
base: iso3098_i
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/italicc.lff:
base: italicc
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/italiccs.lff:
base: italiccs
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/italict.lff:
base: italict
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/kochigothic.lff:
base: kochigothic
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/kochimincho.lff:
base: kochimincho
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/kst32b.lff:
base: kst32b
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/lc_opengost-ar.lff:
base: lc_opengost-ar
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/lc_opengost-br.lff:
base: lc_opengost-br
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/opengosttypea-regular.lff:
base: opengosttypea-regular
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/opengosttypeb-regular.lff:
base: opengosttypeb-regular
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/romanc.lff:
base: romanc
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/romancs.lff:
base: romancs
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/romand.lff:
base: romand
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/romanp.lff:
base: romanp
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/romans.lff:
base: romans
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/romansi.lff:
base: romansi
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/romant.lff:
base: romant
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/scriptc.lff:
base: scriptc
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/scripts.lff:
base: scripts
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/simplex.lff:
base: simplex
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/standard.lff:
base: standard
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/syastro.lff:
base: syastro
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/symap.lff:
base: symap
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/symath.lff:
base: symath
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/symbol.lff:
base: symbol
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/symbol_misc1.lff:
base: symbol_misc1
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/symbol_misc2.lff:
base: symbol_misc2
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/symeteo.lff:
base: symeteo
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/symusic.lff:
base: symusic
font: /tmp/.mount_LibreCKgmbaN/usr/share/librecad/fonts/unicode.lff:
base: unicode
/tmp/.mount_LibreCKgmbaN/usr/share/librecad/patterns
/tmp/.mount_LibreCKgmbaN/usr/share/librecad/qm
QFSFileEngine::open: No file name specified
qt.core.qobject.connect: QObject::connect: No such signal LC_QuickInfoWidget::escape()
qt.core.qobject.connect: QObject::connect:  (sender name:   'LC_QuickInfoWidget')
QLayout: Attempting to add QLayout "" to QG_LibraryWidget "Library", which already has a layout
/tmp/.mount_LibreCKgmbaN/usr/share/librecad/library
/tmp/.mount_LibreCKgmbaN/usr/lib/librecad
Segmentation fault
```

</details>

### Operating System and LibreCAD version info

Version: 2.2.2_alpha1-183-ge26cb0b2<br/>Compiler: GNU GCC 9.4.0<br/>Compiled on: Oct 20 2024<br/>Qt Version: 6.6.3<br/>Boost Version: 1.71.0</br>System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-10-25:
could you please test on the latest AppImage (version 190) - https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.2_alpha-latest?

closed 2024-10-27 by app4soft