LibreCAD/LibreCAD #1852: Metagrid looks ugly: draw Metagrid points as midpoints between Grid points

app4soft, 2024-08-24:
### Expected behavior

Metagrid should looks normal without rendering issues.

### Observed behavior

Actual Metagrid drawing is buggy (something wrong with its line pattern, I guess).

Highlty resommends to draw Metagrid points just as single midpoints between Grid points.

### Steps to reproduce or sample file

Screenshot:

[image]

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-141-g0e0b66e8
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 22 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-08-24:
yes, can confirm this. I've noticed some subtle painting issues (including grid) after migration from QT5 to QT6, I plan to review this a bit later. 
Btw - is the situation the same if "Lines" grid type in "Drawing Preference -> Grid" settings? (actually, grid type should be in Application Settings, that's already in todo list)

app4soft, 2024-08-27:
> Btw - is the situation the same if "Lines" grid type in "Drawing Preference -> Grid" settings?

"Lines" has no issue, except that minor grid lines (each 1/10) shown taller thar major lines.

(But for "Lines" waiting for https://github.com/LibreCAD/LibreCAD/pull/1831 to be merged.)

[image]

[image]

sand1024, 2024-08-27:
> (But for "Lines" waiting for https://github.com/LibreCAD/LibreCAD/pull/1831 to be merged.)
Nope, as far as I know - it's merged already, probably PR is not marked as closed so far. 
BTW - can you play a little with antialiasing setting (in Application Preferences ->Appearance)? Is the result the same with antialiasing on and off?

app4soft, 2024-08-27:
> BTW - can you play a little with antialiasing setting (in Application Preferences ->Appearance)? Is the result the same with antialiasing on and off?

Antialiasing does not matter, even with turned off there is performance issue with rendering text objects.

[simplescreenrecorder-2024-08-27_23.50.24.webm](https://github.com/user-attachments/assets/5c422930-a1f7-4d8d-9338-bd81192cee75)

sand1024, 2024-08-27:
`Is it still possible to build LibreCAD 2.2.2_alpha with Qt4 or Qt5? Or only Qt6 is supported since 2.2.2_alpha?`
Nope, QT6 only. 
Actually, I've also noticed that after migration to QT6 some visual effects occurred so resulting view differs from QT5 version. Most probably I'll review drawing pipeline a bit later.

app4soft, 2024-08-29:
Closing as completed with https://github.com/LibreCAD/LibreCAD/pull/1865

app4soft, 2024-08-29:
JFTR, Actual Metagrid looks better, and almost OK, but could be Metagrid be scaled together with Grid?

Like initially requested, I propose set Metagrid points to shown as 1 between 2 points of Grid, at any level of zoom in/out.

Actually, Metagrid pattern is not scalable and at some zoom levels points of Metagrid coincidents with Grid points, resulting in "glitchy look".

Here is screenshot of latest LibreCAD `2.2.2_alpha`, with 4x zoomed grids and (in yellow rectangle yelow dots is my proposition ofr Metagrid pattern, not that it should not overlap with Grid dots)

[image]

closed 2025-04-14 by u2fly