LibreCAD/LibreCAD #1853: Add Mouse keys support for Shortcuts Mapping

app4soft, 2024-08-24:
### Expected behavior

Shortcuts in Actions Shortcuts Mapping (https://github.com/LibreCAD/LibreCAD/pull/1840) should be fully customizible, and all input devices buttons (keyboard, mouse, pen, etc.) should be fully supported for any action mapping/remapping.

cc @sand1024

### Observed behavior

Actually its impossible to set mouse buttons in Actions Shortcuts Mapping.

### Steps to reproduce or sample file

1.Open "Options > Keyboard Shortuts"
2. Try to use mouse buttons in "Key Sequence" input field.

### Operating System and LibreCAD version info

- Version: 2.2.2_alpha1-141-g0e0b66e8
- Compiler: GNU GCC 9.4.0
- Compiled on: Aug 22 2024
- Qt Version: 6.6.3
- Boost Version: 1.71.0
- System: Debian GNU/Linux 12 (bookworm)

sand1024, 2024-08-24:
well, that's interesting request (it's also possible to think about mouse gestures or so). 
I'll think about it, however, honestly, not sure that it will have high priority. 
Indeed, currently only keyboard shortcuts are supported by the dialog. 
Adding mouse buttons may lead to additional conflicts with existing functionality that heavily relies on mouse and some combinations (mostly SHIFT and CTRL modifiers) has special meaning and are part of actions logic.

app4soft, 2024-08-24:
> Adding mouse buttons may lead to additional conflicts with existing functionality that heavily relies on mouse and some combinations (mostly SHIFT and CTRL modifiers) has special meaning and are part of actions logic.

Make two columns in "Actions Shortcuts Mapping": for keysboard and for input device (for input device allow to set Ctrl/Shift/Alt/other key as modifier key).

| Action name | Keyboard shortcut | Input device |
| === | === | === |
| Zoom In | <kbd>Ctrl</kbd>+<kbd>+</kbd> | <kbd>Scroll Up</kbd> |
| Horizontal Panning | | <kbd>Scroll</kbd>+<kbd>Shift</kbd> |
| Vertical Panning | | <kbd>Scroll</kbd>+<kbd>Ctrl</kbd> |

sand1024, 2026-09-26:
there is a workaround - you may create a custom context menu with single command and assign a shortcut of for it. 

Yet there the invocation is limited to clicks, it's hardly possible that scrolls will be supported.

closed 2026-09-26 by sand1024