LibreCAD/LibreCAD #1856: Line color black shown white in Print Preview - not visible above paper sheet

dellus, 2024-08-26:
Expected behavior: Black lines should be visible in Print Preview.

 Observed behavior: Line color black shown white in Print Preview - not visible above paper sheet.
 Colored lines and images are visible.

Version: 2.2.2_alpha1-143-g466d081d
Compiler: GNU GCC 9.4.0
Compiled on: Aug 25 2024
Qt Version: 6.6.3
Boost Version: 1.71.0
System: fossapup64 9.5
Linux appimage

app4soft, 2024-08-26:
> Line color black shown white in Print Preview

Could you select this line, click "Attributes" and show this dialog?

Also, has you tried click "Toggle Black / White mode" on toolbar in Print preview mode?

[image]

dellus, 2024-08-26:
Toggle B/W mode indeed shows the black lines as black.

"Attributes" of course says "unchanged" for everything.
"Properties" for one of the enties says color "by Layer" and the layer is black/white.
Same behavior if the entity is drawn directly black/white.

app4soft, 2024-08-26:
>  >Could you select this line, click "Attributes" and show this dialog?
> "Attributes" of course says "unchanged" for everything.

Make a screenhsot of this dialog and attach here.

Or maybe you could attach this DXF here? (ZIP it before upload)

dellus, 2024-08-26:
[image]
[image]
[testfile.zip](https://github.com/user-attachments/files/16752897/testfile.zip)

It has nothing to do with a certain drawing, it's the same when drawing on the virgin unnamed document.
It was ok in 2.2.2 alpha1 - 136.

sand1024, 2024-08-26:
thank you for additional info - yes, I've noticed such behavior during debugging one time- yet was not not able to reproduce it after several attempts so decided that it was just a side effect of debugger... 

Yet it seems that there is some issue with pen setting for preview (actually, there is additional logic to prevent drawing entities with the same color as background one). 

I'll check for this, hope with test file it will be simpler to reproduce.

app4soft, 2024-08-27:
> I'll check for this, hope with test file it will be simpler to reproduce.

Its reproducible with any file, even with a new created, in latest LibreCAD `2.2.2_alpha`.

Set layer color to "Black / White" and pen color to "By Layer".

app4soft, 2024-08-31:
Issue could be closed as it fixed already.

dellus, 2024-10-18:
The issue is back in 2.2.2 alpha1 181 in Linux appimage.
Version: 2.2.2_alpha1-181-g8b3cf9ca<br/>Compiler: GNU GCC 9.4.0<br/>Compiled on: Oct 17 2024<br/>Qt Version: 6.6.3<br/>Boost Version: 1.71.0
System: fossapup64 9.5

dellus, 2024-10-25:
Fixed again in 2.2.2 alpha1 190

closed 2024-10-25 by dellus