LibreCAD/LibreCAD #1859: MText R2L transposes numeral

KoTCiB, 2024-08-26:
### Expected behavior
When you use R2L MText it should not transpose numeric values.
For example, assume you want to enter an address as 'Main 123'
'naiM 123' + R2L should result in 'Main 123'

Numerals are read L2R (at least in Hebrew) and should also be kept in the current placement in the sentence.

### Observed behavior
The result for the above is '321 Main' as numeral are treated as other characters.

### Steps to reproduce or sample file

### Operating System and LibreCAD version info
Version: 2.2.1_rc3-16-g0a0a81de
Compiler: GNU GCC 12.2.0
Compiled on: Aug 23 2024
Qt Version: 5.15.2
Boost Version: 1.75.0
System: Windows 10 Version 2009

dxli, 2024-08-26:
This is more complicated than expected.

Chinese can be R2L, including numbers in Chinese. I am not clear on Arabic digits in Chinese when writing right to left. Chinese can be also top to bottom. Need to consider a way to support generic directions.

KoTCiB, 2024-08-27:
Another thing to consider is compatibility.  I almost always convert my final LibreCAD DXF file to DWG as it is shared with AutoCAD users.  I do not have an AutoCAD license and use the Open Design Alliance converter.  Thus, as long as I am familiar with LibreCAD behavior, I can work around it to achieve a readable DWG file.
For example, I work with a Hebrew outline font.  If I use Text, the DWG file shows question marks for all characters.  This may be an ODA converter quirk.  On the other hand, the same text in MText (L2R or R2L) is displayed correctly after conversion.
Thank you very much

closed 2026-10-09 by dxli