LibreCAD/LibreCAD #1860: Command line field loses focus if escape out of current command

Questions2000, 2024-08-27:
LibreCAD-2.2.2_alpha1-98-ga3bb1b64.AppImage
Linux / Debian 12 based distro / GTK based DE
QT5 5.15.8

I might be overlooking something to why this occurs but,
I noticed that if you run a command such as "line" or "trim"
if you activate the command and then instantly hit your "escape" key as if you have changed your mind you did not want to run that command, then the focus is not given back to the command line field.
As long as would not cause any conflicts, would anyone be willing to make it so the command line field gets the focus back after the "escape" key is hit?

Thank You

sand1024, 2024-08-27:
in general, it will be great to review the entire support of keyboard on the application-wise level. 
Of course, there is separate command for focusing cmd widget, yet indeed it might require an additional key press. 
I'll take a look on this.

Questions2000, 2024-08-31:
@sand1024 
Thank you for being willing to look into this.
Not sure if this will help you in any way but I also forgot to mention that the right mouse button acts in the way that I am mentioning for the Escape key to function.
If apply a command and then hit your RMB in the drawing pane, it cancels the command fine and gives focus back to the command line field.
Anyways, hopefully this is helpful in some way.
Thank you again for your help

Questions2000, 2024-10-02:
Not sure how much this would require to fix but could be another item that could be closed in regards to the "issues to close" 1902 list

Questions2000, 2024-10-27:
I tested this out in the latest AppImage release and this still occurs.
I have updated the original post above with new information and more examples if anyone is interested.

RenateUSB, 2024-11-09:
Absolutely! This is very annoying. How can you have the input focus go **nowhere**?
l `SP` 0,0 `SP` 1,1 `SP` `ESC` `ESC`

Questions2000, 2025-01-18:
Thank you for agreeing.

If anyone is interested, I was able to run this in terminal using -d6 option to get some more information about maybe what is happening.
I will make two comments below
-This one shows the actual issue that occurs using "Escape" key and CL field loses focus.
-The one after this comment will be example of using RMB in drawing pane to cancel the action which does apply focus back to the CL field

HERE I ACTIVATED THE "LI" COMMAND TO DRAW A LINE
(THIS IS THE INFORMATION SHOWN AT THE BOTTOM, NOT REALLY RELEVANT)
```
RS_ActionDrawLine::RS_ActionDrawLine
RS_EventHandler::setCurrentAction
RS_EventHandler::setCurrentAction Draw lines
RS_EventHandler::setCurrentAction: init current action
RS_ActionDrawLine::init
RS_ActionDrawLine::reset
RS_EventHandler::setCurrentAction: show options
RS_EventHandler::setCurrentAction: set predecessor
RS_EventHandler::setCurrentAction: cleaning up..
RS_EventHandler::cleanUp
RS_Math::derationalize: expr = '2'
RS_EventHandler::cleanUp: OK
RS_EventHandler::setCurrentAction: debugging actions
---
Current
Action 000: Draw lines [active]
RS_GraphicView::setCurrentAction: OK
QG_ActionHandler::setCurrentAction(): OK
QG_ActionHandler::command: current action set
```

AFTER THE ABOVE, THIS IS THE INFO SHOWN AFTER I HIT "ESCAPE" KEY TO CANCEL THE ACTION
(The IGNORED items at the bottom are where I am trying to type in a new command but the command line no longer gets focus)
```
QC_ApplicationWindow::slotWindowActivated begin
QG_ActionHandler::command: escape
call action Draw lines
RS_ActionDrawLine::init
RS_ActionInterface::finish
RS_ActionInterface::finish: OK
RS_ActionDrawLine::reset
RS_EventHandler::cleanUp
RS_Math::derationalize: expr = '2'
RS_EventHandler::cleanUp: OK
QG_ActionHandler::command: back
QC_ApplicationWindow::KeyPressEvent: IGNORED
QC_ApplicationWindow::KeyPressEvent: IGNORED
QC_ApplicationWindow::KeyPressEvent: IGNORED

Questions2000, 2025-01-18:
Here is the example of using RMB to cancel the action, which works fine, this applies focus back to the CL field as expected.  Providing this because maybe something can be seen why this method does work and the "Escape" key issue mentioned above does not.

HERE I ACTIVATED THE "LI" COMMAND TO DRAW A LINE
(THIS IS THE INFORMATION SHOWN AT THE BOTTOM, NOT REALLY RELEVANT)
```
RS_ActionDrawLine::RS_ActionDrawLine
RS_EventHandler::setCurrentAction
RS_EventHandler::setCurrentAction Draw lines
RS_EventHandler::setCurrentAction: init current action
RS_ActionDrawLine::init
RS_ActionDrawLine::reset
RS_EventHandler::setCurrentAction: show options
RS_EventHandler::setCurrentAction: set predecessor
RS_EventHandler::setCurrentAction: cleaning up..
RS_EventHandler::cleanUp
RS_Math::derationalize: expr = '2'
RS_EventHandler::cleanUp: OK
RS_EventHandler::setCurrentAction: debugging actions
---
Current
Action 000: Draw lines [active]
RS_GraphicView::setCurrentAction: OK
QG_ActionHandler::setCurrentAction(): OK
QG_ActionHandler::command: current action set
```

HERE IS WHERE I BRING MY MOUSE CURSOR INTO THE DRAWING PANE AND HIT RMB TO CANCEL THE LINE COMMAND
```
call action Draw lines
RS_ActionDrawLine::init
RS_ActionInterface::finish
RS_ActionInterface::finish: OK
RS_ActionDrawLine::reset
RS_EventHandler::cleanUp
RS_Math::derationalize: expr = '2'
RS_EventHandler::cleanUp: OK
QG_GraphicView::mouseReleaseEvent: OK
RS_Snapper::catchEntity
```
Hopefully some of this information will be helpful.

Questions2000, 2025-03-21:
I was just wondering if anything can be seen in the information I have provided above that could solve this issue?
Thanks

sand1024, 2025-03-22:
well, nothing so far, thank you. yet I'll ask if some additional clarifications will be needed. 
Hope it will be fixed soon, but most probably it will a change together with update of cmd, so far there are several other priority things in the queue, so let the issue be open for some time.